goshante/ats20_ats_ex #2: Tuning to band pattern not working (AM 9 or 10, SW 0 or 5)

hzprky, 2024-03-28:
First of all thank you very much for your work and wonderfull firmware for that little radio. It absolutely improves the overall usability of the radio.

One observation, not really sure if you would call it a bug...

On SW the tunig pattern is 0 or 5 kHz, so almost every station is either on xxxx0 or xxxx5 kHz. If you start tuning from let's say 6072kHz upwards and the chosen tuning step width is 5 kHz, the next reasonable frequency up would be 6075 kHz (down 6070 kHz). However, the firmware moves to 6077 (6067 down), so simply adding (subtracting) the step width.

I'm not sure if this is a simple fix - i had a look at the code and assume it is located in the main module ATS_EX.ino in the function "void doFrequencyTune()". The handling would need to be individual for each band an inside HAM bands it would be even different, as there are no discrete steps agreed for the sending stations, depending on the operator. On AM it is depending on Europe, Americas and Asia if the tunig step width is 9 or 10kHz. 

I hope you can understand my approach and have an idea to change (not necessarily fix) it.

Thanks again for your work!

Xian

goshante, 2024-03-29:
Thank you for your feedback!
There is no band pattern provided in this firmware. It is designed to make incremental steps and scanning also works like that.

It is also not possible to improve because firmware reaches the maximum image size limit of ATMega328p flash memory. So no major improvements (such as band patterns) will be added. It will add extra 100-300 bytes to image size if i implement band pattern logic for LW/MW/SW and add an option to the settings for this feature (Europe/USA). 
In v1.15 binary image takes 31600~ KB of 31722 KB maximum. If i add or remove some code compiler will break byte code alignment and image will be 31800 KB. And it's even without some new feature.
It is already optimeized to be minimal size as possible.

closed 2024-03-29 by goshante

hzprky, 2024-03-31:
thanks for your quick reply! However, i will try to think about an optimized minimal solutionfor the challenge ;-). Will let you know, if i find something of value.