goshante/ats20_ats_ex #4: error loading Hex

spirit89, 2024-03-29:
Ok everything was working great until the second  V1.14 and up. The first V1.14 worked and all down.
The first V1.14 March 26 worked. I went back to GitHub there was second V1.14 March 27 on loading that file
error and the am/fm scan and some other function not working. went back to V1.13 loaded, and everything works.
This is the best program for the ats20. Thank you Dave

goshante, 2024-03-29:
Try 1.15

spirit89, 2024-03-29:
Ok 1.15 still gives error loading and the scan same as 1.14. loaded 1.13 back on no errors everything works. My radio has nano
on a board off of the main board.

spirit89, 2024-04-01:
Tried USB 3.0 and dummy firmware now it loads 1.15. But now have auto scan problem like the one you closed. scans to the first station. push it again you can hear it scanning but not changing display and then the audio's back to the first station with display. manual tuning works. If I go back to 1.13 scanning ok. Thanks for all the help..

goshante, 2024-04-01:
@spirit89 
I fixed stucked scan in v1.16, you could try it now
If it stuck again just press it once more or make few steps forward

closed 2024-04-03 by goshante