goshante/ats20_ats_ex #6: Additional features for ATS 20+

Amenra78, 2024-04-04:
Hello and welcome. Would it be possible to add a very useful squelch function to the firmware and scan all bands? I am writing on behalf of myself and several dozen users of this fantastic firmware. Regards

goshante, 2024-04-04:
Please, read "About new features requests" section in README

Amenra78, 2024-04-04:
I understand, I just thought that you could do it at the expense of, say, the MW and LW bands. I must admit that only a small handful of users use it. But I understand your suggestion. I'm sorry and have a nice day.

goshante, 2024-04-04:
Squelch is not possible due to very rough values of RSSI, there is no receivers on Si4732 with squelch function. There is little possibility it could work on CB band, but there is not enough space.

However, source code is open and you can try implement scanning switch automatically between bands if you want to.

closed 2024-04-04 by goshante