goshante/ats20_ats_ex #7: ATS20+

celadoor, 2024-04-16:
This works fine on my ATS20+ but the battery divider is already installed to pin A1 on ATS20+ and not pin A2, changed in _defs.h_  to _"#define BATTERY_VOLTAGE_PIN A1"_ and the charging indicator works now, you might want to mention this in the readme.

SV2HZT, 2024-08-11:
To reverse Rotary Encoder direction in arduino IDE line 235 should be  g_encoderCount = (encoderStatus == DIR_CW) ? -1 : 1;

G8PTN, 2024-08-20:
Hi,
Many thanks to Vincent for this excellent update for the ATS-20/ATS-20+. The feature in SSB using the BFO to provide a smooth tuning experience over a 32 kHz span is a great addition.

It would be nice to have a remote capability, maybe with an internal BT/BLE to serial module. I did disable the RDS feature, which freed up enough program memory to allow for a simple volume up/down, frequency step up/down and output of the tuned frequency (g_currentFrequency and g_currentBFO). I guess if a remote option is required, it would be better to have a separate version supporting a standard radio protocol, allowing use of existing PC control programs and/or a Phone/Tablet App. 

For the ATS-20+, I also changed the definition for the “BATTERY_VOLTAGE_PIN” to “A1” instead of the default “A2”.
The default settings for the rotary encoder worked correctly on the ATS-20+. If the direction is wrong, it would perhaps be better to make the change in the defs.h file, rather than the main code.

For programming  using the Mini-USB connector, I made a VBUS isolator to avoid any conflict between the 5V (BUS) and the 3V3 regulator on the ATS-20+.

The ATS-20+ version I received appeared to have the 115200 baud rate bootloader on the Arduino Nano.

[image]

G8PTN, 2024-08-25:
Hi,
As an experiment, I managed to create a very basic Bluetooth interface with a simple Android App.

Hardware:
Using a JDY31 Bluetooth module with a few additional components.
The firmware can also be uploaded over Bluetooth with the Arduino IDE.

[image]

Firmware:
Minor modification to add compile directives to reduce program memory to allow for remote additions.

[image]

Added code to send information out over the serial interface to the BT module.

[image]

Added code to extract serial data from the BT module and mimic the buttons and rotary encoder.

[image]

Android App:
Created using MIT App Inventor.
This recreates most of the display information and allows very basic control.

[image]

[image]

73, Dave

G8PTN, 2024-08-25:
Hi,
A minor updated. I added three additional flags over the ATS-20+ to BT serial interface to allow the remote GUI to show when parameters are in change mode.

PS: Apologies for the rather large screenshots on the previous post, I should have reduced the size down before uploading.
73, Dave

[image]

[image]

G8PTN, 2024-08-30:
Hi,
A minor point on the battery meter.
The ATmega328 ADC is using the ‘AVCC’ pin for Vref. This is connected to the ‘+5V’ on the Arduino Nano, which is connected to the ATS-20 3V3 LDO regulator output as shown below).

[image]

The voltage calculation is based on V_BAT = 2 * [ADC * Vref/1024]
The firmware assumes that Vref is fixed at 3.3V.
This is valid for V_BAT >= 3.3V + (LDO drop-out).
At lower battery voltages, the reading will be invalid. On the ATS-20+ that I have, the reported battery percentage does not fall below 17%, corresponding to a battery voltage of approximately 3.4V.

[image]

It would be possible to use either the internal band gap reference (1.1V) or the 'AREF' pin (with external reference). This would require adjusting voltage divider feeding the analogue 'A1' pin. The potential divider resistors are under the Nano, so it would be easier to switch to a different analogue pin.

Since the the battering monitoring is accurate down to 3.4V, no modifications are proposed at the moment.
73, Dave

tormoz108, 2024-10-29:
Буду писать по русски, извините.
Действительно, замерить разряд батареи ниже 17% не получается. На этой цифре процессор зависает.
Решений данной проблемы может быть несколько
1. Установить уровень разряда = 0 на 3.4 В. До этого напряжения работают стабилизаторы напряжения. Далее стоит принудительно понизить чаcтоту процессора, чтобы продлить агонию.
2. Использовать в качестве опорного напряжения внутренний источник 1.1В. Но для этого придется менять коэффициент делителя напряжения батареи. Тогда можно будет мерить напряжение честно, но на 3.2В процессор зависнет все равно. Значимого ухудшения приема, когда стабилизаторы превращаются в перемычки я не заметил.
3. Измерять  напряжение питания процессора , и зная это значение, использовать его как опороное напряжения для измерения батареи. Для этого вход АЦП подключается к источнику стабильного напряжения 1.1В  и оно измеряется относительно неизвестного VCC. https://arduino.ru/forum/programmirovanie/izmerenie-vcc. Я слепил такую прошивку, это возможно, но имеет скорее развлекательную ценность. Отъело дополнительно порядка 2% памяти. Измеряются два уровня - батареи и VCC. Показало, что после снижения до 2.3 вольта и ниже значения напряжения батареи и питания процессора совпали полностью.

vatonek, 2025-05-23:
NOTE to measuring battery on ATS20 Plus:

As voltage V_BAT/2 is already on pin A1, simplest hardware modification is just connect A2 to A1. Unfortunately, I am not sure how ATS_EX configures pin A1 I/O logic. If both pins A1 and A2 are configured as analog input, the modification will work.

It is possible also simple software solution:
measure voltage on both pins A1 and A2 and then use the pin where voltage detected. And modify corresponding part of manual, of course ...