goshante/ats20_ats_ex #15: Band list.

Luciano77ARG, 2024-08-12:
Vincent I want to thank you for the excellent work you did with the swling.ru firmware! 
I'm asking you about the list of bands that the originals firmware had with this format:

_Band band[] = {
    {FM_BAND_TYPE, 8700, 10800, 9830, 3, 0, "FM"},   // FM 87-108 MHZ; default 98.3; step 3; bw auto
    {LW_BAND_TYPE, 100, 520, 300, 1, 4, "LW"},         // LW
    {MW_BAND_TYPE, 520, 1720, 810, 3, 5, "MW"},        // AM/MW
    {SW_BAND_TYPE, 1700, 2000, 1850, 0, 5, "160M"},    // 160 meters HAM
    {SW_BAND_TYPE, 2000, 3200, 2250, 1, 4, "120SW"},   // 120 meters BC
    {SW_BAND_TYPE, 3200, 3500, 3300, 1, 4, "90SW"},    // 90 meters BC_
.....

I would like to change the limits of each band.
The name of each band is interesting, for example 40M or 49SW

Thanks!

rogerclarkmelbourne, 2024-09-27:
Try looking here

https://github.com/goshante/ats20_ats_ex/blob/8fd52cf73496fd3bdf0c7660bb317c44c70d75fd/ATS_EX/globals.h#L243-L272

HG9FS, 2025-07-11:
...
     /* LF */ { 30, 300, 225, 0, 4 }, 
     /* HF */ { 300, 3000, 540, 3, 4 }, 
     /* HF */ { 3000, 30000, 3000, 0, 4 }, 
     /* UHF */ { 30000000, 300000000, 87500000, 1, 0 }, 
  ...