goshante/ats20_ats_ex #19: Locations to solder for the battery remaining display!

Sarah-C, 2024-09-16:
If we solder to the negative and positive of the battery, there's always some current used even when the device is off.
So I looked on the board for a suitable spot to use that's part of the circuit within the on/off switch control.

I found the 5V pad shown in the screenshots was straight from the battery through the power switch. 
So I heat shrank (to avoid any shorts from the resistor bare wire legs) two 10Kh resistors and soldered one to this pad, and one to the ground pin of the Arduino Nano - simply because it's easy to solder to.

Erm.... a mistake! I cut the resistor leg a bit short for the 5v pin resistor, so the yellow wire you see from A2 curving around is to correct the short resistor leg! It's just soldered at the point between the two resistors that are joined.

To be fair though, it works out a bit better as soldering a single solid wire to A2 is easier than two resistor legs twisted together. It leaves room on either side.

I hope someone finds the pictures useful, and perhaps Vincent will put them in some instructions somewhere?

[image]

[image]

[image]

[image]

Sarah-C, 2024-09-19:
Blimey! That's good advice Dave!

I think I'll recompile, and remove the extra soldering. Thanks for letting everyone here know!

Sarah-C, 2024-09-19:
Update!

My soldering has been taken out, and the potential divider on pin A1 is working fine.

As Vincent said - if you have Visual Studio, load the .SLN file.....
If you have the Arduino IDE, simply goto File/Open/and find the **ATS_EX.ino** file in the project you download.
The other files will appear in several other tabs.
Click on the defs.h tab, and make the changes David G8PTN talks about.

It'll give some warnings about ANSI C++ data type conversions when you compile it, but they can be ignored.

It'll upload just fine, and the battery display will show the power left! Much less work than soldering.

kirbimir, 2024-09-19:
Hi,
I follow for lasst few days your topic and collect info about upgrading ATS-20+ which should arrive to me within nex 10 days or so. I have never flashed to ATS radio and I familiarized  myelf with all info I could so procedures are  relatively clear to me . I  prepared all files from ARDUINO IDE  v 2.3.3 (nightly build)  - photo of files is here. 
However, can you  **[image]**  by means of photo  (screen capture) or  so ? Any advice is welcome!
I am afraid I will enter data incorrectly and ruin the ATS-20+  .
Thank you in advance, with best regards Zarko

[image]

Sarah-C, 2024-09-19:
Ahhhh! I used the older Arduino 1.8.19.......

I have one question kirbimir- the folders you have.... did you install them into your Arduino IDE?

The convention is to open the "Library Manager" in the Arduino IDE, and search for the libraries in there. You won't see the folders themselves.
To me it looks like you've downloaded the folders directly - this should be fine when you build (when the folders are all in the right place!) It'll still work - but you won't get the benefit of the libraries auto-updating for bug fixes and new features.

You can click the "Tick" button, to see if it compiles... you don't need to have the device yet!
If you can't get the compile and you get errors for missing libraries - it's best to go the library manager and search for them, and download them in there. The Arduino IDE will link all the libraries it's managing up to the compiled Radio program you're making. You don't need to fiddle with getting compile paths correct.

Don't worry about "bricking" your radio doing this........ the Arduino Nano has a small program in it that doesn't change. The "Bootloader"...... this ALWAYS checks to see if some PC program is trying to update the device software before the device runs your program. It only pauses for a fraction of a second to check.....
Importantly!
Your program you upload WONT overwrite the bootloader, so if the sketch for some reason doesn't work, you'll always get to upload again, perhaps a different sketch. But it should go fine.

If you have problems, you can use the AVRDUDESS with the ready-compiled HEX file. It'll upload the pre-compiled code without the Arduino/Visual Studio IDE needed.
 

I've included a picture of what you should see for the libraries.......
[image]

Image showing the tab that contains the #Definition what needs editing.
[image]

kirbimir, 2024-09-19:
Hi,
Thank you sso much for your detailed explanations. I got  what is  in my directory by clicking "install" in Library Manager in ARDUINO IDE. Do not know are these usefull or not. However , I found one .hex file   ATS_EX_v 1.18.hex by someone  giving youtube video of update instructions. **Is this the file to use in AVRDUDESS ?** 
Anyway, I will study your comments to get better understanding.
Thank you again, Zarko

G8PTN, 2024-09-19:
Hi,
I also used Arduino IDE 1.18.19, primarily because it allows for a portable installation, which keeps everything in a known area on the PC. Yes, there are several compile warnings in Arduino IDE, but everything seems to work okay.

A couple of minor points:
- Uploading the hex file using the ISP interface will erase the flash (including the bootloader area), so it is best to do the updates via the USB interface as stated by Sarah, since the bootloader will be maintained.
- When using the USB interface, it is best to isolate the 5V connection if possible, since this can conflict with the 3V3 that is feeding the Arduino Nano (see link).
https://github.com/goshante/ats20_ats_ex/issues/7#issuecomment-2298663399

73, Dave

kirbimir, 2024-09-20:
I studied in details ARDUINO IDE v.2.3.3  user instructions they have put on., and tried several examples and procedures. I think it iss quite clear now to me how to do upgrade on my coming ATS-20+  box. When I do the upgrade using procedure decribed in intruction I wil post the photos of upgraded box.Thank you all for your king comments! 
One thing I do not understand : how to isolate v5 described here ? I cannot see from photo what it is . Can you give more detalis? Thank you.
Best regards
Kirbimir

G8PTN, 2024-09-20:
Hi,
To isolate the VBUS (5V) on the USB interface, others have just cut the 5V wire in the USB cable.

I built a small isolator breadboard with USB-A and Micro-USB sockets that I had available.
The GND, D+ and D- are connected between the associated pins on the two sockets.
The VBUS (5V) is connected between the two sockets via a break link.

To isolate the VBUS (5V), I remove the link and use two un-modified USB cables.
Cable 1:	USB-A from PC to Micro-USB on isolator board.
Cable 2:	USB-A from isolator board to Mini-USB on ATS-20+.

I hope that helps.
73, Dave

[image]

[image]

kirbimir, 2024-09-21:
Hi Dave,

Thanks for additional explanation.  However, could you post a picture of complete connection or diagram how to connect this gadget to laptop and AT-20+ ?

Looks to me it needs some miniature soldering activity.

I must confess it is not quite clear to me,

Thank you
Kirbimir

G8PTN, 2024-09-21:
Hi Kirbimir,
There is no miniature soldering required, the USB breakout boards are all on 0.1 inch (2.5mm) pitch.
As stated previously, others have just cut a cable in half and joined all the wires apart from the 5V. This is the easiest option.
Others have not even isolated the 5V, and in most cases it still works, but there is a risk that something could be damaged.
Comments from other groups attached.
https://groups.io/g/si47xx/message/625
73, Dave

[image]

kirbimir, 2024-09-21:
Thank you Dave for additional information. What I see on the photo the micro female port on ATS-20+ must be connected to 5V box  and home wall to give pover to the device  or to laptop ? 
Therefore , mini USB port with cut 5V cable in the cable itself is only for data transfer. Do I get this corectly?
Regards, Kirbimir

G8PTN, 2024-09-21:
Hi Kirbimir,
No quite right.

The Nano was designed to operate from 5V, but on the ATS-20+ the interface to other devices needs to be 3.3V, so they obviously decided to run the Nano from the 3V3. This is okay until you connect a USB cable from a PC. The diode that is in the VBUS (5V) path on the Nano will try and pull the internal supply voltage above 3.3V, which could cause other devices on the ATS-20+ to be damaged.

By cutting the 5V from the PC to the ATS-20 USB port, it  prevents this from happening. You do ned to have the ATS-20+ switched ON when downloading via the USB interface.

The ATS-20+ could have been designed to have level translation on the internal signals and there would have been no issues on the 5V from the USB connector.

You are correct that the USB connector is now only be used for data transfer, which needs "D+", "D-" and "GND" connections.
73, Dave

kirbimir, 2024-09-21:
Hi Dave,
 Thanks for explanation. Now,simply :  **do I connect both cables** namely, micro port USB to laptop  **and** also mini USB with cut wire 5V ?

Thanks for reply , Kirbimir

G8PTN, 2024-09-21:
Hi Kirbimir,
No, if you cut the wire in the Mini-USB cable, you only need to use that cable.
I only built the adapter so that I could use standard unmodified cables.
73, Dave

kirbimir, 2024-09-27:
Hi,
to-day I recieved AT-20+ box . No problems I detected . I post few photos. Now, I ask : what to do  next ? I see on desplay V3.0.4.   I prepared mini  data cable as per Dave's instructions namely  cutting v5 red wire - photo.
Does box need any update via ARDUINO  IDE v 2.3.3  and how to proceed? Thanks for advice!
Anyway , I wouild like to soldied  2x 10 kOhm resistores as per Dave's photo to get battery  showing use. Any additional advice on this issue?
Thank you , 
Kirbimir

[image]
[image]
[image]
[image]
[image]

G8PTN, 2024-09-27:
Hi Kirbimir,
You have two options. Both can be done via the “Mini-USB” connector on the back of the ATS-20+ receiver.
Option 1:	Upload the hex file using AVRDUDESS.
Option 2:	Modify the code in the Arduino IDE.

I hope this helps you,
73, Dave

**OPTION 1**
If you are not confident using “Option 2”, I have created an updated hex file with the “A2” to “A1” pin swap.
This may be the easiest option for you.
ATS_EX_v1.18.hex		Original. Needs hardware modification to connect potential divider to Nano "A2" pin.
ATS_EX_v1_18_A1.hex	“A2” to “A1” swap. No hardware change required.
I have put both of these hex files in the attached ATX_EX_hex.zip.
[ATX_EX_hex.zip](https://github.com/user-attachments/files/17169239/ATX_EX_hex.zip)

• Install and run AVRDUDESS.
• Under the “Presets” option select “Arduino Uno (ATmega328P)".
• Power on the ATS-20+ and connect the modified USB cable to the computer.
• Select the appropriate COM port.
• In the “Flash” option, select the hex file.
• Select the “Program!” tab. The hex file will be uploaded.
• The ATS-20+ should reboot. You may need to do an EEPROM reset after upload (see below).

Notes:
- The ATS_EX_v1_18_A1.hex only applies to the ATS-20+ which has the connection to the "A1" pin of the Nano.
- Do not upload this file via the ISP connector, it will override the bootloader. The file is for the "ATmega328P".
- To reset the EEPROM, hold the rotary encoder pushed whilst turning on the ATS-20+
- Example AVRDUDESS screenshot below with ATS_EX_v1_18_A1.hex selected.

[image]

**OPTION 2**
The method of compiling the code in Arduino IDE 2.x was covered in previous messages from Sarah.
I did not use IDE 2.x, but these are the basic steps.

• Run the Arduino IDE
• Install the following libraries using the “Library Manager”.
   PU2CLR_SI4735
   Tiny4kOLED
   TinyOLED-Fonts
• From File/Open select the ATS_EX.ino file,
• This should load the sketch and all associated files.
• In the defs.h
   Change from:       #define BATTERY_VOLTAGE_PIN A2
   Change to: 	        #define BATTERY_VOLTAGE_PIN A1 
• In the  “Tools” option, check the "Board" and "Processor" settings.
   Board:		Arduino Nano
   Processor:	ATmega328P
• From the “Sketch” option, select “Verify/Compile”.
   There may be  some warnings, as identified by Sarah, but these should not cause a problem.
• Power on the ATS-20+ and connect the modified USB cable to the computer.
• From the “Tools” option, check that the Port is set correctly.
• From the “Sketch” option, select the “Upload” option and the sketch will Verify/Compile and upload to the ATS-20+
• The ATS-20+ should reboot, You may need to do an EEPROM reset after upload (see below).

Notes:
- The change only applies to the ATS-20+ which has the connection to the "A1" pin of the Nano.
- To reset the EEPROM, hold the rotary encoder pushed whilst turning on the ATS-20+

kirbimir, 2024-09-28:
Hi Dave,
Thank you for detailed explenations. However, I have 2 problems:
**Option 1 :** 
- I have Arduino nano but in your AVRDUDESS screenshot is stated Uno
- Will upload of ATS_EX_v1_18_A1.hex   **using modified mini USB** 
   cable override the bootloader ?
**Option 2 :**
I did everything as stated but :
in Arduino IDE From File/Open I cannot select the ATS_EX.ino file . It does not show in "libraries" . Only :
PU2CLR_SI4735
Tiny4kOLED
TinyOLED-Fonts
show.
How to proceed ?

**Questions :**
Will simply soldering resistors without Options 1 or  2  to the box as it is now  give the result ?
**Can I simply soldier in the box as it is now as per :**

[image]

or 

[image]

I do not understand : " The ATS_EX_v1_18_A1.hex only applies to the ATS-20+ **which has the connection to the "A1**" pin of the Nano." 

I appologize for constant questioning !

Kirbimir

G8PTN, 2024-09-28:
Hi Kirbimir,
Option 1:
I had already updated the instructions to select the Arduino Nano.
You just have to remember to set the baud rate to 115200 as shown in the instructions.
If you upload via the USB Mini-USB connector on the ATS-20+, it will **NOT** override the bootloader.
If you use the ATS_EX_v1_18_A1.hex file, you do **NOT** need to make any hardware changes.
There is already a connection on the ATS-20_ PCB from the switched battery via potential divider resistors to Nano pin “A1”.
[image]

Option 2:
You do not use the hex file directly with the IDE. You need to compile the sketch and then use the “Upload” option as stated in the instructions.

_Questions :
Will simply soldering resistors without Options 1 or 2 to the box as it is now give the result ?_
Option 1: ATS_EX_v1_18.hex WOULD require the resistors and wiring.
Option 1: ATS_EX_v1_18_A1.hex does NOT require the resistors ore wires.
Option 2: Does NOT require hardware changes.

_Can I simply soldier in the box as it is now as per :_
No,

kirbimir, 2024-09-28:
Hi Dave,

I get this :
[image]

Whats wrong?

G8PTN, 2024-09-28:
Hi Kirbimir,
Not sure why you see the "red" message. All the settings look correct.
Give it a try, it should not break anything.
[image]

73, Dave

kirbimir, 2024-09-28:
Hi Dave,Ž
At that time  field  EEPROM was filled with data from "Flash" . Do nok know how I entered this. When comparing with your sample I noticed it should be empty. So I deleted contend and now it is  showing blue.  Obviously my mistake.
I will now push Publish! 
Kirbimir

kirbimir, 2024-09-28:
[image]

No success.
How to solve this  ?

G8PTN, 2024-09-28:
Hi Kirbimir,
It looks like it is not seeing the radio.
Did you check that COM port is correct in device manager?
Is the radio turned ON?

G8PTN, 2024-09-28:
When you modified the cable, did you check that all connections are okay, apart from the 5V?

G8PTN, 2024-09-28:
You should see something like this in device manager.
[image]

kirbimir, 2024-09-28:
Well, I checked the cable . It should ne OK .
I device Mngr  shows :

[image]

G8PTN, 2024-09-28:
You should not need to change the speed, AVRDUDESS will make the setting.
What is the device on COM3?
Are you sure the radio is COM4?
If the radio is COM3, you may need to get the CH340 driver.
73, Dave

G8PTN, 2024-09-28:
Hi Kirbimir,
Unfortunately, I will be off-line until later.
Hopefully you will be able to get the COM port working, and then it should upload correctly.
I will check in later.
73, Dave

kirbimir, 2024-09-28:
I pulled out mini usb and in Device mngr  COM4  dissapeared. I inserted  mini cable COM 4  appeared. So I judge box in on COM4.

kirbimir, 2024-09-28:
Hi Dave,

I succeded to make flash  under settings : 

[image]

However, after some errors  it was done . I presed  rotary knob anhd powered device . 
Now it shows new display I give pictures:

[image]

[image]

[image]

However, it can not switch  to FM . Ony AM  shows although some volume changes.
I do not know to make shown settings.

At the bottom  right it shows now 92% . Is that battery indicator ? 

How to get FM back?

or 

how to reset to original setting.
 at least I found that COM 4 is working .

Thanks
Kirbimir

G8PTN, 2024-09-28:
Hi Kirbimir,
Excellent, well done.

It looks like your ATS-20+ had a slightly different bootloader.
Some are set for a baud rate of 115200 and others 57600, so it it not a problem.

- Yes, the indicator on the bottom right hand side of the display is the battery indicator.

-  You will still need to move the switch at the back to FM/AM(SSB) depending on the frequency you are tuned to, since it selects the Antenna input on the SI4732 receiver chip.

- This is not my software, I am just one of the many users of it.

- There are lots of features and they are all covered on the main GitHub page.
https://github.com/goshante/ats20_ats_ex?tab=readme-ov-file#user-manual

I hope you have fun with the ATS-20+
73, Dave

kirbimir, 2024-10-04:
Hi Dave,
First I want to thank you again for your help. AT-20+ works very fine and I have lots of fun with it. 
Now I ask you if perhaps you can  figure out what is wrong with my ham radio Anytzone AT778uv . It can not communicate with CHIRP or Annytone factory program application . I was thinking perhaps Prolific chip program cable I got with radio  was to blame so I made my own cable . Both cables are visible in Program Manger . However, my cable shpws a bit different. Error messeges I get are " no initial response from  radio" & "cennot comunicate " , " read data failed ". It is my first such radio and I do not know how to solve this. I tried to find solutions on internet but all solutions are about cable not visible in Device Mngr. and similar. 
If you can perhaps give me some opinion is this error something to do with radio itself,perhaps?
I enclose some photos.
Thanks in advance 
Kirbimir

[image]
[image]
[image]
[image]
[image]
[image]

G8PTN, 2024-10-04:
Hi Kirbimir,
The information you have shown appears to be for the AT-779UV not the AT-778UV.
I have never owned either of these, but looking at information on-line, it seems that the serial data interface for the AT-778UV is a 1-wire type, where the TX and RX data is shared.
If your radio is the AT-778UV, you could perhaps try the attached circuit, where a diode is used to combine the TXD and RXD signals on the USB to Serial adapter.
https://on3jt.byze.be/programming-cable-for-anytone-at-778uv 
73, Dave

kirbimir, 2024-10-06:
Hi Dave,
Thank you for excellent advice ! I made new interface according to schech you attached in your last reply (photos  I apologice for big unedited photos). 
Obviouisly trick is in installing a diode from RX pin to TX pin ! **BUT** I only succeded in download radio data using both Anytobe software & CHIRP **after experimenting with COM4 advanced configurations** (succesful configuration is as per photos) .
Result file from CHIRP is empty beacuse I prior download radio data made Reset all in radio so it is empty. Now I need to uploda  new data from some template  in CHIRP.
Perhaps this  solution will be of help to many users of AT778uv because searching on internet I find many "how to" questions on this.
Thank you very much again!
With best regards,
Kirbimir
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

G8PTN, 2024-10-06:
Hi Kirbimir,
I am pleased it now seems to be working.
Yes, Ideally there should be an open-drain or open-collector buffer, but the diode does a close to equivalent job, although the logic ‘0’ will be a diode drop above GND.
ICOM use a similar 1-wire interface called “CI-V”.
73, Dave

h0bbl3s, 2025-01-02:
https://github.com/goshante/ats20_ats_ex/issues/19#issuecomment-2361906649

I just wanted to say I've been following this repo for a while as I'm working on a firmware of my own, and your help (everyone but especially G8PTN) has been invaluable. 

For all of the folks with the nano version I wanted to say this is what I use rather than mess up a cable to cut out the 5v.
 
[image]

[It can be found on amazon here](https://a.co/d/dLv981z)

kirbimir, 2025-01-20:
Hi Dave,

Since yesterday I can not tune into  stations onb my ATS-20+ . Nothing I did to cause this. 
I tried to fash again the .hex file with AVRDUDE but could not . Error show could not connect to COM port. and not in sync . 
Is there some hard reset button inside the machine . What can I do to to correct this assuming the machine itself did not go broke?

kirbimir, 2025-01-20:
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x03
Failed chip erase: uploading error: exit status 1

G8PTN, 2025-01-20:
Hi Kirbimir,
When you say it does not tune, does it still boot up and show the information on the screen?
Does it respond to any of the button presses?

The screenshot shows that there is no response from the Nano on the serial connection, the Nano may have failed for some reason.
I assume you are using the USB cable with the 5V isolated.
73, Dave

kirbimir, 2025-01-20:
I can tun it on/off  and it shows the initial screen  ok , but I find now I can tune only to some 105.50  MHz and listen to this radio station. All other previous stations do not work. Also I see no difference when switching AM/FM  switch. 
Have no  idea what happened I didn't mess with anything . 
Kirbimir

kirbimir, 2025-01-20:
I used isolated cable as before

G8PTN, 2025-01-20:
Hi Kirbimir,
Okay, it seems that the Nana is not completely failed, so I don’t know why it can’t be programmed.
The receiver part does not appear to be correct.
Did you try resetting the EEPROM to see if it makes any change?
To reset the EEPROM push and hold the rotary encoder button whilst turning the receiver on.
73, Dave

kirbimir, 2025-01-20:
I did several times but no change.

G8PTN, 2025-01-20:
Hi Kirbimir,
With respect to the Mini-USB interface, does the port show up on the computer and is the serial baud rate set correctly?

It seems like the Nano is still working, but the SI4732 receiver chip has gone deaf.
This could be due to accidental static discharge to the antenna or from overload from some very strong nearby transmitter.

Perhaps check that RF connector to see if there are any open or short circuits.
Unfortunately, I don’t really have any other suggestions.
Sorry I can’t solve the problem for you this time.
73, Dave

G8PTN, 2025-01-20:
Perhaps also check that the 3.3V supply voltage.
[image]

73, Dave

G8PTN, 2025-01-20:
Hi Kirbimir,
One final thought.

There are two 3.3V supply rails, one for the Nano and one for the SI4732 receiver.
If possible, check that both supplies are at the correct voltage.

[image]

[image]

73, Dave

kirbimir, 2025-01-20:
I checked 3V pin to GND it gives 3.3V  on DMM

G8PTN, 2025-01-20:
Hi Kirbimir,
I have the ATS-25 AMP, and I would not really recommend it. It has issues with the power supply circuit, and it generates quite a lot of RFI from the ESP32 and TFT screen.

There are many different versions of the ATS-25, the later ones may have some improvements. One version has an in-built mixer circuit to allow Airband (VHF) AM reception.

All of the ATS receivers, ATS-20, ATS-25, ATS-120 use exactly the same SI4732 receiver chip, so fundamentally the performance is going to be similar. Some models may have better front-end filtering.

If you are interested in the CW, RTTY and FT8 decoding, then the ATS-25 is quite good. The firmware is not open source, so there are no options for experimentation.

In summary, I think the ATS-20+ offers the same RF performance as the ATS-25 at a much lower cost, but the decoding features of the ATS-25 are an obvious advantage.

The ATS family of receivers seem to be constantly changing, probably to encourage people to purchase the newer version.
73, Dave