goshante/ats20_ats_ex #20: Battery voltage sensing hardware modification suggestion

rogerclarkmelbourne, 2024-09-28:
Hi

Your instructions for where to attach the battery sensing hardware could cause problems if the radio was not used for a long time.

Specifically the instruction to connect directly to the battery will cause the battery to be slowly flattened

_other directly to the positive of the lithium battery_

I know that the current drain will only be 3.7 / 20k = 185uA, but over many weeks this will cause the battery to go completely flat and may cause it to no longer be usable.

e.g.  If the battery is 200mAH then 185uA would flatten it in 1081 hours = 45 days.

A better solution is connect the resistor divider to the switched side of the supply, then once the radio is turned off there will be no current drain from the battery and the radio could be unused for several years before the battery became naturally flat.

On my version of the ATS-20, which has the Nano daughter board, I found some unused pads in the middle of the PCB which have the switched battery voltage supplied to them, and I used one of these pads

G8PTN, 2024-09-28:
Hi,
Yes, it is better to use the switched battery connection.
If you have the ATS-20+ version the resistors and connections are already on the PCB.
https://github.com/goshante/ats20_ats_ex/issues/19#issue-2528048684
73, Dave

rogerclarkmelbourne, 2024-09-28:
Arrrghh ..  Sorry.

I didn't notice that other issue. 

Not that it matters, but my PCB is different from the one shown in that issue.

I'll close this as its  duplicate.

PS.

I've also noticed another small annoyance.  When the radio is first turned on, the first screen date always displays the battery as 8%.

I think there is a redundant screen update when the radio turns on, which displays things before it should do.

I'm going to look at the code later and see if its fixable

rogerclarkmelbourne, 2024-09-28:
LOL.

I just noticed that the battery thing is also mentioned in

https://github.com/goshante/ats20_ats_ex/issues/8

rogerclarkmelbourne, 2024-09-28:
Looks like this repo is not actively monitored

closed 2024-09-28 by rogerclarkmelbourne

G8PTN, 2024-09-28:
Hi,
The initial low battery percentage is probably because the firmware does an average to minimise constant changing of the displayed reading.

I think the repo is monitored, but the author is busy on other commitments and does not plan updates at the moment.

73, Dave

rogerclarkmelbourne, 2024-09-28:
OK. 

It looks like after the splash screen is cleared, there are 2 updates to the display within about 100ms

I'm trying to build the code now to investigate

Re: Battery level averging

Normaly the initial battery average should be set to the instantaenous value on the ADC unless its known to be noisy. 

Alternatively the splash screen is held for several seconds, and this would give plenty of time to average the ADC

rogerclarkmelbourne, 2024-09-28:
BTW. Do you know if there is any way to get rid of the annoying multiple clicks when the radio is turned on ?

G8PTN, 2024-09-28:
Hi,
Yes, the initial value could be set. I am not sure it does that.
You will also find that in practice the % will not fall below 17~18 percent, even if the battery is near empty (see link).
https://github.com/goshante/ats20_ats_ex/issues/7#issuecomment-2321081548

With respect to the audio clicks, it is a function of the SI4732 and audio amplifier.
During configuration of the SI4732, the DC level on the ROUT/LOUT changes which causes the clicks.
You will get the clicks when changing band, switching mode etc..
It is common on most of the radios that use this chip.
Some designs add an external circuit, but I have not investigated this further.