goshante/ats20_ats_ex #21: ATS20+ always reboot (linux)

wipedlifepotato, 2024-10-17:
I try to flash though avrdude but my ats20+ always reboot while this process going. I also tried AVRDUDESS though mono with the binutils-avr package and avrdude but my ats20+ just reboot and all what going

G8PTN, 2024-10-17:
Hi,
Are you trying to flash the ATS20+ via the ISP interface or USB?

If flashing via USB.
- The “Presets” setting in AVRDUDESS should be set to “Arduino Nano (ATmega328P)”.
   Some bootloaders are using 57600 baud and others 115200 baud.
- There is a possible conflict between the external USB (VBUS = 5V) and the internal voltage rail 3V3 (see attached thread).
   https://github.com/goshante/ats20_ats_ex/issues/19#issue-2528048684

If it is failing to program, I would expect a message to be reported in AVRDUDESS.
73, Dave

wipedlifepotato, 2024-10-22:
Trouble is baud rate

closed 2024-10-22 by wipedlifepotato