goshante/ats20_ats_ex #23: Can you say what is the best ats 20 plus settings for the best reception in any situation the best settings

ShasaLegend, 2024-10-29:


G8PTN, 2024-10-29:
Hi,
In my experience, the default settings seem to work well.
I did change ATT from “AUT” to “1”, which removes the slight background noise in SSB modes. This is covered in the manual and does not really need to be changed.

[image]

All the configuration options are available for the user to adjust if required. There will not necessarily be a single optimum setting for all conditions.
73, Dave

ShasaLegend, 2024-10-30:
yes i know the default settings are good but with ATT i have problems 
but the SM,SSM,AVC is better to set the default or change it?
in setting SM stay 0
SSM stay SNR
AVC stay 60

ShasaLegend, 2024-10-30:
yes i know the default settings are good but with ATT i have problems 
but the SM,SSM,AVC is better to set the default or change it?
in setting SM stay 0
SSM stay SNR
AVC stay 60
and IF signal comes out from ats?

G8PTN, 2024-10-30:
Hi,
I would expect that the default settings used in the Goshante firmware match those recommended by the PU2CLR SI473x library.

There is no IF output from the ATS-20. The demodulation is done digitally in the SI473x receiver chip.

As previously stated, the optimum settings will probably vary depending on band, mode, antenna, user preference etc.

ATT
Increasing the attenuation level might be useful under strong signals conditions, but there is no front-end filtering so overload from out of band signals will always be a problem.
 
SM		Soft Mute
SSM		SSB Soft Mute
The soft mute only really affects the tuning experience. This was mainly noticeable in SSB.
With the Goshante firmware it is much less of a problem because of the tuning method used.
[image]

AVC		Automatic Volume Control
SVC		Enable AVC for SSB
The AVC might be useful for weak signals, but it will also increase the level of the noise.
73, Dave

G8PTN, 2024-10-30:
Hi,
If AVC set to 60 works better for you, then leave it at that.
73, Dave

ShasaLegend, 2024-10-30:
ok thanks have a nice day

closed 2024-11-08 by ShasaLegend