goshante/ats20_ats_ex #25: Hi is it possible to make a mini update of that firmware in the update, I would like that when the Search function is turned off in the settings, then when you press the button on the encoder not to change the step, but to change the BFO, it will be more convenient that way.
Thanks

ShasaLegend, 2024-11-10:


G8PTN, 2024-11-10:
Hi,
My understanding is that the author of the firmware (goshante) is busy on other commitments and does not plan updates at the moment.

When tuning in SSB mode, the tuning mechanism automatically uses the BFO as described below.
[image]

The BFO adjustment in the settings menu is to allow an offset to be made for calibration so that the receivers displayed frequency matches a known frequency standard.
[image]

73, Dave

ShasaLegend, 2024-11-11:
Hi yes i understand but it was better when is SSB or the encoder button to change BFO
but if you have the sketch in ino can you please send i want to change that code for me to modify.
Thanks

G8PTN, 2024-11-11:
Hi,
The code is already available in the github.
https://github.com/goshante/ats20_ats_ex/tree/master/ATS_EX
73, Dave

ShasaLegend, 2024-11-13:
what is that library font14×24sevenSeg.h?

G8PTN, 2024-11-13:
Hi,
I assume it is used for the frequency display as stated in the user manual.
73, Dave

[image]

tormoz108, 2024-11-14:
tmp

tormoz108, 2024-11-14:
Гораздо лучше, чем енять шаг настройки сделать его переменным. При быстром вращении крутилки *10 при замедлении - возвращать к нормальному. При смене направления вращения - так же возвращать к норме. Для переключения использовать гистерезис.

tormoz108, 2024-11-14:
uint16_t midEconderTikTime;
bool g_lastseekDirection;    // куда крутили до этого
byte n;                  // коэффициент ускорения кручения
byte swithD;               //  туда сюда

void doFrequencyTune()
{
  if (g_displayOn)
  {

    g_seekDirection = g_encoderCount == 1 ? 1 : 0;
    //  if ( EconderTikTime > 270) midEconderTikTime += 20;
    EconderTikTime = (millis() - g_lastFreqChange);   //sid
    //  midEconderTikTime = midEconderTikTime>>1 +  midEconderTikTime>>2 + EconderTikTime>>2;  // сглаживающий фильтр
    midEconderTikTime = midEconderTikTime / 2 +  midEconderTikTime / 4 + EconderTikTime / 4; // сглаживающий фильтр
    //   if (EconderTikTime < 250)

    if (midEconderTikTime < 200) n = 10;   //ускорение крутилки

    if ( midEconderTikTime > 350) n = 1;

    if ( g_lastseekDirection != g_seekDirection)
    {
      n = 1;
      midEconderTikTime = 500;
    }

    if (g_currentMode == FM)
    {
      g_currentFrequency += g_tabStepFM[g_FMStepIndex] * g_encoderCount

G8PTN, 2024-11-14:
Hi,
Yes, accelerated tuning can be advantageous.
There are probably several different options that could be added, but many are perhaps user specific.
I decided to add Bluetooth control, so that I can try customising options via an App.
Good luck with any changes.
73, Dave
[image]

ShasaLegend, 2024-11-15:
for me i just want waterfall and dbm sensibility show and that's it, can you please show me how to make that app wia cable connection

G8PTN, 2024-11-15:
Hi,
I have no plans to release the code. It only provides basic serial monitoring/control.
The principle was shown in this thread.
https://github.com/goshante/ats20_ats_ex/issues/7#issuecomment-2308964134
If you want to have a waterfall, I suggest you look at something like the ATS-25, which uses the same radio IC, but has a more powerful MCU.
73, Dave

closed 2024-11-22 by ShasaLegend