goshante/ats20_ats_ex #26: BFO Offset Calibration

RFCanada, 2024-11-17:
Hello,

First, I just want to say Thank You for your firmware. The best display!

I have noticed a few possible "bugs/issues?" with current version v1.18 running on my ATS-20.

On SSB, my frequency readout is off by 2kHz. Using BFO Offset Calibration is not helping as it's limited to 60/-60. Can this value be made larger? Say 250/-250? This would cure my display not showing correct issue. My ATS-20 has always been like this from new, on any firmware.

When on SSB with S-Meter enabled, the audio becomes slightly distorted. Is this normal?

I am not sure what Sync does? When enabled on SSB, when stations are off frequency, is it supposed to "auto tune" them? It is not doing that for me. Please advise.

I love the look of the display, it's very easy on the eyes vs. other firmware available which tends to overcrowd the small display. The battery mod works excellent and seems to be fairly accurate. Overall great work!

Regards,

-Rob

G8PTN, 2024-11-17:
Hi Rob,
Hopefully, the firmware author will be able to respond to your questions.
I have made a few notes below.
Regards, Dave

Frequency and BFO:
I have not noticed a significant frequency error on the ATS-20+ that I have. The readout is within +/- 200Hz without any calibration (BFO) adjustment. I guess the 32.768kHz crystal used by the radio IC may be a little more off on your device.

The BFO configuration parameter is handled has an 8-bit signed integer (-128 to +127).
[image]
The “doSwitchLogic” function, which auto wraps will work correctly for -127 to +126.
In addition, the display has 3 characters allocated so it will not show correctly for values lower than “-99”.
(e.g. -100 would show as “-00”, -101 would shows as “-01”).

In summary the simple 1-line change would allow +/-99 = +/- 990Hz, but it would not solve your problem.
It should be possible to modify the code to allow for a greater range, but more changes would be required.

S-Meter:
When the S-Meter is enabled, there are frequent I2C serial bus accesses to the radio IC and OLED display. Unfortunately, this can cause noise on the received signal.

A general observation (not related to the S-Meter noise) that has been covered in the manual, is there is a general background noise when the ATT = AUT. I generally leave it set at “1”.
[image]

Sync (AFC):
I have not really experimented with this option and there is little information about it in the on-line documentation relating to the SI473x radio IC.
https://pu2clr.github.io/SI4735/extras/apidoc/html/group__group17.html

RFCanada, 2024-11-18:
Hi Dave,

Thanks for your response, much appreciated.

If the +/-99 Hz is very easy to do, then by all means it makes sense to implement the change. If +/-250 Hz is doable with the remaining memory, then of course that makes sense also. I do notice weird performance when using Sync on SSB on my ATS-20, so I don't enable it much. Once I am dialled into a SSB station, it is relatively stable with minimal drift.

Thanks for clarifying the S-Meter noise. I only use it on AM/FM bands anyway, so I don't hear the slight distortion. 

I have changed my ATT = AUT to a setting of 1, thanks for the tip.

A 2.4 kHz SSB filter would have been nice. 1.8, 2.4 and 3 kHz for SSB would be all that is required. Overall, for the price range and fun of playing around with the little SDR receiver, it's worth every penny.

I have the ATS-20+ with all-in-one circuit board (1024 model) and ATS-25 AIR on order, so looking forward to playing around with something new.

Regards,

-Rob

RFCanada, 2024-11-19:
Hi Dave,

I took your suggestions and recompiled the code from the ATS_EX.ino file.

I was able to change the value from * 10 to * 25 without issue, and now the BFO Offset Calibration works for me with a larger range.

About NFM, most of the FM repeaters in Canada are WFM still. I do not believe Canada is mandated to switch to NFM like the US.

Thank you for all your help!

-Rob

G8PTN, 2024-11-19:
Hi Rob,
That's great, glad to help.
On the ATS-25 AIR, I should perhaps have said on AIR/VHF is can only do the same modes as the HF section of the receiver, AM, USB, LSB, CW. It basically uses a down converter.
73, Dave