goshante/ats20_ats_ex #28: Hi

ShasaLegend, 2024-11-22:
Does Si4732 have a IFIN?
i want to use a panadapter to my ats 20 plus

G8PTN, 2024-11-22:
Hi,
I am not sure if you mean IFOUT?
There is no IF output from the SI473x.
[image]

73, Dave

ShasaLegend, 2024-11-22:
Hi,i mean IFIN where if brut in entering in the si4732 i want to make a panadapter for ats

G8PTN, 2024-11-22:
Hi,
Normally a panadapter needs a tap from the IF stage of the receiver, which can not be done with the Si4732.
You could use an RF splitter and something like an RTL-SDR dongle, but you may as well then use the dongle as the receiver rather than the ATS-20.
Please could you show a diagram of what you are trying to achieve.
73, Dave

ShasaLegend, 2024-12-06:
no i dont have a diagram or a circuit i want to try but if it is not IFin i will not try

closed 2024-12-19 by ShasaLegend