goshante/ats20_ats_ex #30: Reverse Rotary Encoder direction

Angelus8, 2024-12-21:
Hello,
Thx for the firmware that I've just implemented on my ATS20 (USB C version).

I'd like to reverse the rotary encoder direction like it was suggested by SV2HZT
"To reverse Rotary Encoder direction in arduino IDE line 235 should be g_encoderCount = (encoderStatus == DIR_CW) ? -1 : 1;"

Can someone explain me how to modify the .hex to  re-flash my device with the proper values ?

Regards
Angel

Angelus8, 2024-12-21:
Ok I've downloaded "ATS_EX.ino" from here 
https://github.com/goshante/ats20_ats_ex/blob/master/ATS_EX/ATS_EX.ino
Then modified the line with notepad to change values. 

But now I don't know how to make this become an .hex file...

Angelus8, 2024-12-21:
So I've downloaded Arduino IDE 2.3.4
Created a folder named ATS_EX inside which is ATS_EX.ino
Now I can open the .ino and change the values...

How do I recompile this now ???

Angelus8, 2024-12-21:
Compilation error: SI4735.h: No such file or directory

Angelus8, 2024-12-21:
It works !
There were just libraries missing !
I followed this tuto
https://github.com/patomoi/ATS-20-Starter-Pack

    

> download release 
>     extract **...(and modifiy as suggested above for the rotary encoder to behave well)...**
>     install CH341 Driver
>     open Arduino ide and **update libraries (pu2clr/SI4735, Tiny4kOLED, TinyOLED-Fonts)**
>     open sketch of you interest (the examples included here [with-without bat mod] or the original included in library -update: this sketch is now in the library exemples-) **...(and modifiy as suggested above for the rotary encoder to behave well)...**
>     verify and upload
>     if a memory issue happens, go to bandtable section and comment (//) the bands that you not use ( pick 1 FM band and 1 MW band )
> 

I'm very happy to have helped myself :)

Issue solved hope this will help newbies as me !

Merry Christmas

AlainTH, 2024-12-30:
Hello folks.
Any chance someone would share the modified .hex file for reversed encoder?
Thanks in advance, and Season's Greetings to all.

Angelus8, 2024-12-30:
Hello
I don't know how to compile the Hex file to reverse the encoder but uploading the modded file on the ATS is easy with Arduino IDE. I've followed this tuto https://github.com/patomoi/ATS-20-Starter-Pack

To summarise
1.      Install CH341 driver and look in the device manager (right click) if your radio is COM 3 or COM 4 ... when you connect it.
2.      Install and open Arduino IDE
          1. In Tools/Manage Libraries Search and install (pu2clr/SI4735, Tiny4kOLED, TinyOLED-Fonts)
          2. In Tools/Board Choose your board (Mine is Uno) like the manual said (https://github.com/goshante/ats20_ats_ex)
             - Identify your board as described here There are two types of ATS-20 receivers: Those on Arduino Nano and those on
             - Arduino Uno. They are quite easy to  distinguish. You need to disassemble... This is important to know before flashing...
          3. In Tools/Port choose the port COM 3,4,etc... that you've seen in the device manager
4.      Download the release file (https://github.com/goshante/ats20_ats_ex/archive/refs/tags/v1.18.zip) and unzip it
5.      In Arduino IDE/File/Open Open ATS_EX.ino from the zip file you've unzipped
         Then In the ATS_EX.ino tab modify line 235 from "g_encoderCount = (encoderStatus == DIR_CW) ? **1 : -1**;" to **-1:1**
6.     On the upper left corner click Verify button (V) then power ON your Radio, when it is ON click on Upload (the right Arrow)
7.     Reset EEPROM memory. To do this just hold the Encoder Button while turning receiver ON.

Enjoy
Angel

AlainTH, 2024-12-31:
Thanks. The .hex file is found in a subfolder of %TEMP% after compilation.
I wanted to avoid going through the installation of the Arduino IDE and learning it. Hence my asking for the .hex file.
My ATS20+ is one of the batches with a defective USB COM port, so I have to use an USBasp programmer. Your scenario still mostly applies anyway.
I'll go with the IDE then. 
All the best
Alain

Angelus8, 2024-12-31:
https://github.com/goshante/ats20_ats_ex/releases/tag/v1.18

In the source code zip you've got all the necessary so when you open ATS_EX.ino from the unpacked archive all the dependencies are open together. You can see the tabs and modify the code you want. Then if you've updated the 3 libraries and chosen the right platform and the right COM port, all should be ok.

The .hex and .zip formats aren't supported...

Alain sounds like a Francophone name... Peut être pouvons nous échanger en Français ;)
Angel

Angelus8, 2024-12-31:
Here it is 
[ATS_EX.ino.hex.zip](https://github.com/user-attachments/files/18281831/ATS_EX.ino.hex.zip)

Hope this is what you're looking for
Angel

AlainTH, 2024-12-31:
Merci. Oui nous pourrions échanger en Français, mais pour que tout le monde puisse profiter de ces échanges, c'est sans doute mieux de rester en Anglais.

I think that I have everything set up correctly. The sketch compiles with quite a few errors, none fatal: redefinitions, ISO C++ not compliant code, default arguments etc. Ends up with:
```
Sketch uses 30708 bytes (99%) of program storage space. Maximum is 30720 bytes.
Global variables use 1245 bytes (60%) of dynamic memory, leaving 803 bytes for local variables. Maximum is 2048 bytes.
```
Do you get the same values?

Just saw your download link. I'll check against mine. Thanks again.

I think I might be better opening a new issue for this discrepancy in hex files instead of polluting this one.

Angelus8, 2024-12-31:
I've got this

> Le croquis utilise 30708 octets (95%) de l'espace de stockage de programmes. Le maximum est de 32256 octets.
Les variables globales utilisent 1245 octets (60%) de mémoire dynamique, ce qui laisse 803 octets pour les variables locales. Le maximum est de 2048 octets.

The only difference is the 95% vs 99% you have... Maybe a memory issue ???

Yes opening a new issue should be a better choice.

@+
Angel

AlainTH, 2024-12-31:
> I've got this
> 
> > Le croquis utilise 30708 octets (95%) de l'espace de stockage de programmes. Le maximum est de 32256 octets.
> > Les variables globales utilisent 1245 octets (60%) de mémoire dynamique, ce qui laisse 803 octets pour les variables locales. Le maximum est de 2048 octets.
> 
> The only difference is the 95% vs 99% you have... Maybe a memory issue ??? EDIT or the bootloader ?

No. That's because you've probably set Board = Arduino Uno instead of Arduino Nano. I just checked with that setting and I now get the same message as yours (language set aside):
_Sketch uses 30708 bytes (95%) of program storage space. Maximum is 32256 bytes._

My ATS-20+ definitely has an Arduino Nano. I don't know about yours and I don't know how critical it is to have it set correctly.

Merci pour ces échanges et bonne année
Alain

G8PTN, 2025-01-03:
Hi Alain,
Please see information below with respect to the ATmega328P Uno and Nano available program space.
73, Dave

**Summary**
Sorry if the information below sounds a bit complicated. The main point to note is that if you want to gain the extra 1536 bytes of flash for program space, you would need to compile for the Uno board and make sure that the BOOTSZ(1:0) fuses bits are set accordingly.

In the Arduino environment when using a programmer connected to the ISP, the fuse bits get set automatically. This can also be done in AVRDUDESS by selecting the appropriate options.

**Notes**
Although the Arduino IDE uses the same optiboot_atmega328.hex for the bootloader (which is 512 bytes, 256 words) for Uno and Nano boards, the BOOTSZ(1:0) fuse bits are set to different values.
[image]

[image]

[image]

Hence the number of bytes available for program code space will be different as you have seen from the compiles.
[image]

I believe the reason for the Nano being set to a bootloader size of 1024 words is to maintain backwards compatibility with the “Old Bootloader”.

Even though the Nano allocates 1024 words (2048 bytes) for bootloader, the smaller bootloader image still works. This is because when the bootloader is installed, the flash is erased (0xFF), and although it will boot from an address based on 1024 words, the 0xFF value will appear as “NOP” and it will advance until it reaches the optiboot code.

In the Arduino IDE environment when flashing via the ISP using a programmer, it will automatically set the BOOTSZ(1:0) fuse bits for either the Uno or Nano, based on the board selection.

If the Nano module is shipped with a bootloader, the BOOTSZ(1:0) fuses bits are likely to be set for a Nano and therefore compiling for an Uno and flashing the hex file would not actually free up additional space for the program without setting the BOOTSZ(1:0) fuse bits accordingly. This can be done in AVRDUDESS by selecting the appropriate options.