goshante/ats20_ats_ex #31: Where to solder resistors ATS 20 new version (single USB C)

Angelus8, 2024-12-23:
Hello,
This is the ATS 20 new version (single USB C)

[image]

It upgraded to this new firmware like a charm by the USB C, with the charging cable provided. Obviously changing A2 pin to A1 in the sketch does nothing cause it is not the + version. The battery icon still shows 0%. This is what the board looks like.

[image]

Does anyone know where to solder my resistors ?
Regards
Angel

Angelus8, 2024-12-24:
Mine is an Arduino Uno. anyway, I can't figure where is the A1 pin ?

G8PTN, 2024-12-24:
Hi,
I don’t know if all unused pins have been connected for the “U3” dual footprint, I suspect not.
From the datasheet, the location for “ADC1” and “ADC2” pins for each package is shown below.

[image]

Assuming the unused pins are not tracked, pin “ADC2” on the 28-pin package may be the easiest to solder.
73, Dave

Angelus8, 2024-12-28:
Hello
I did this

[image]

[image]

And this

[image]

But unfortunately I still got this

[image]

Angelus8, 2024-12-28:
May I link the pins ?

Angelus8, 2024-12-28:
Here is a better view

[image]

Angelus8, 2024-12-28:
On the A1 pin the radio is always ON wen the battery is plugged in...

[image]

I'll try on A2 pin... Much more hard to link...

G8PTN, 2024-12-28:
Hi,
Looking at the PCB, it does not look like all traces have been tracked between the two footprints.

If you could try and connect the junction of the potential divider resistors to the ADC2 (corner pin) on the 28-pin package and then change the defs.h to “A2”. Since the two footprint are not fully tracked out, you can connect the corner pin to you existing arrangement.

[image]

Perhaps also check the voltage on the potential divider with a DVM, it should be the battery voltage divided by 2.
Unfortunately, I don’t think there are any other unused ADC pins that can be swapped.

73, Dave

Angelus8, 2024-12-28:
> connect the junction of the potential divider resistors to the ADC2 and then change the defs.h to “A2"

Unfortunately doing this gives no reading and the ON/OFF button is not shutting the radio anymore. (I had inverted A2 and A1
in the previous message... I've just edited to correct)

So I tried on The A1 pin 

[image]
 
Plugging the battery, switch ON/OFF on the OFF position, the ATS boots and shows 36% then 18% then 0% when I put it ON... So the ATS remains always on... I'll check the solder of the resistor on the switch...

Angelus8, 2024-12-28:
PYJAMAS !!!!

Since the beginning I systematically connected the battery in the HP connector instead of the Battery connector... How stupid am I !
I just figured out... Now that everything is de soldered !!! 
The radio works fine so I'm at the same point... 
So I'm going to try again with A2 which is way easier to link...

G8PTN, 2024-12-28:
Hi,
All understood. I guess that is the danger when both connectors are the same,
I hope it has not damaged the speaker amplifier chip.
73, Dave

Angelus8, 2024-12-28:
Youpi !

All is ok ! I've the battery reading !
I used pin A1

[image]

And now I can see the battery level...

[image]

Thank you for your help G8PTN. It was a long journey :)

G8PTN, 2024-12-28:
Hi,
Excellent, glad to help.
73, Dave

kirbimir, 2025-01-23:
Hi Dave,

After attempts to flash .hex file to  ATS-20+  in Win11 it crossed my mind to try  avrdudess in my Linux Debian 12  disk.
I succeded to get avrdudess in linux and made the flash of  ATS_EX_v1_18_A1
The resulted message is :

[image]

Do you have any idea about this? 

Regards 
Kirbimir

kirbimir, 2025-01-23:
[image]

I changed  hex file  . this is now result 

Kirbimir

G8PTN, 2025-01-23:
Hi Kirbimir,
I have not used the Linux version of AVRDUDESS.
Did the second attempt solve the problem on the ATS20+ or is it still not receiving correctly?

Looking at the log, you had something selected differently in AVRDUDESS.
For the first file you had “**flash:v**”, which is verify mode.
“v” = read the specified device memory and the specified file and perform a verify operation
This will have failed because when it checks the complete flash memory against the file contents, it includes the bootloader area. When programming via USB (TX/RX data), the bootloader  is not touched and so I would probably expect to see a different.

For the second file you had “**flash:w**”, which is write mode.
“w” = read the specified file and write it to the specified device memory.

If you try the ATS_EX_v1_18_A1.hex with the AVRDUDESS in “write” mode, I expect it will program correctly.
73, Dave

kirbimir, 2025-01-23:
Hi Dave,

I made 4 scenarios and uploaded hex fles  
But some 4  FM stations returned not all .

Obviously Windows is making effort to block Chinese chip in ATS-20+   but in linux avrduddes  works . I had the same problem with ham radio AT778uv  and windows 11 only accepted conenction to radio after I made new cable with new chip .

However, in this case it seems to me  hex A file is corrupted somehow.
Can you post here  correct hex A file I could download ?
I ordered new ATS-2+ and I will program it again in linux. Hope it will work !

Regards 
Kirbimir

[image]

[image]

[image]

[image]

G8PTN, 2025-01-23:
Hi Kirbimir,

The location shown being different in AVRDUDESS is in the bootloader section and I would expect that to be the case.

As stated previously, when writing via USB(TXD/RXD), the bootloader will not be erased or overwritten, which is the safest option for you to use. Therefore the bootloader section will be the original as programmed by the supplier of the ATS20+.

For completeness I have created a new zip with 3 files.
[ATX_EX_230125.zip](https://github.com/user-attachments/files/18524648/ATX_EX_230125.zip)

- ATS_EX_v1.18.hex
With bootloader code.
Original. Needs hardware modification to connect potential divider to Nano "A2" pin.

- **ATS_EX_v1_18_A1.hex**
No bootloader code.
“A2” to “A1” swap. No hardware change required.

- ATS_EX_v1_18_A1_with_bootloader.hex
With bootloader code.
“A2” to “A1” swap. No hardware change required.
Note: This was the file I had previously sent, and I renamed it to ATS_EX_v1_18_A1.hex.

Try the new file “**ATS_EX_v1_18_A1.hex**”, which does not contain the bootloader section.

Perhaps AVRDUDES is behaving differently in Linux, and it checks all locations that appear in the hex file or it previously just flagged a warning only.

I hope this all makes sense and sorry if I have confused you.

73, Dave

kirbimir, 2025-01-24:
Hi,
I flashed all hex files.one with bootloader reported to be not verified.
Other two worked ok. Can you mail me hex file that will erase everything on
chip including maybe bootloader? I made reset on the button inside but I
see not difference.Kirbimir

G8PTN, 2025-01-24:
Hi Kirbimir,

The result is as expected. It shows that the verify after writing checks all locations contained in the file with what it reads back from the device. Its slightly strange in that I would expect the Windows version to behave in the same way. There is an option in AVRDUDESS to disable verification after writing "Disable verify (-V)", but I would not recommend setting this, since all locations would not be checked.

If you want to erase the bootloader you need to do it via the ISP connector on the Nano board. To do this you need either a USBASP programmer or another Arduino board setup to be a programmer. I would not recommend going down this route unless you need to, since if something goes wrong, you can put the device into a non-recoverable state. It is only necessary to use the ISP connector method if the device does not contain a bootloader. The original devices do not have the bootloader code, but generally all modules including Uno, Nano and those delivered with units such as the ATS-20 will have been programmed with a bootloader.

The files that you have already contains the bootloader code.
- ATS_EX_v1.18.hex
With bootloader code.
If programmed via the ISP, this sets the bootloader baud rate to 57600.

- ATS_EX_v1_18_A1_with_bootloader.hex
With bootloader code.
If programmed via the ISP, this sets the bootloader baud rate to 115200.

73, Dave

kirbimir, 2025-01-24:
Thanks Dave for your help. New ATS-20+  should come aprrox. 10 Feb.  Can you istruct me which hex file to  upload   when  it comes to get the  upgraded performance? 
With best regards
Kirbimir

G8PTN, 2025-01-24:
Hi Kirbimir,

The only difference in functionality is the battery meter pin swap. Assuming the new ATS-20+ is the same as the one you have, you should be able to use the **ATS_EX_v1_18_A1.hex** from the zip that I sent.

Remember, depending on which bootloader has been loaded by the seller, you may have to set the baud rate to either 57600 or 115200 in AVRDUDESS.

I hope the new receiver works okay.
73, Dave

kirbimir, 2025-02-05:
Hi Dave, 
I recieved yesterday new ATS-20+  buit unfortunatelly it was not recognized nither in Win11 nor Linux ports using modified cable with cut 5V wire.
However, after pluging it in laptop  with regular cable it works only with this cable plugged in .
Good news I got tota immediate refund from ALLIEXORESS .
Old reciever I could program in Win11 with avrdudess and your hex A1 file but it  recieves only few FM stations.

Any idea about above ?
Regards
Kirbimir

[image]
[image]

G8PTN, 2025-02-05:
Hi Kirbimir,
As previously stated, the old receiver sounds like it has a fault on the RF side with low sensitivity.
On the new receiver, assuming it is using the same hardware, did you try programming it with AVRDUDESS?
You may need to try the different baud rate 57600 or 115200.
73, Dave

G8PTN, 2025-02-05:
Hi Kirbimir,
With the isolated cable the radio would need to be turned on, otherwise the Nano board would not be powered.
73, Dave

kirbimir, 2025-02-05:
> From "G8PTN" ***@***.***>
> To "goshante/ats20_ats_ex" ***@***.***>
> Cc "Kirbimir" ***@***.***>; "Comment"
> Date 25/02/05 6:09:05 PM
> Subject Re: [goshante/ats20_ats_ex] Where to solder resistors ATS 20 new
> version (single USB C) (Issue #31)

Wit isolated cable it does NOT turn  ON at all. With standar cable it 
turns ON.

This email has been checked for viruses by Avast antivirus software.
www.avast.com

G8PTN, 2025-02-05:
Hi Kirbimir,
Does the radio turn on when you don't have the isolated USB cable connected?
The isolated cable is just using the USB+/- and GND, so I am not sure why it would stop the radio turning on?
73, Dave

kirbimir, 2025-02-05:
Without standard cable  (no cable at all) it does NOT turn ON. 
Regards,
Kirbimir

G8PTN, 2025-02-05:
Hi,
Without the USB cable connected, can you check to see if there is 3V3 on the Nano pins when the power switch is on.
I assume you have charged the battery?
If the battery is charged, the on board 3V3 regulator may have been damaged by connecting with a USB cable without isolation.
73, Dave

kirbimir, 2025-02-05:
I ll do that chevcke and get back .

G8PTN, 2025-02-05:
Hi Kirbimir,
Yes, that is exactly why the 5V should be cut on the USB cable.

The voltage should be **3.3V** and it is supplied by the ATS-20+ on-board 3V3 regulator.

Unfortunately, connecting the non-isolated cable may have damaged the Nano and/or the 3V3 regulator.

[image]

Without the isolated 5V cable, the 5V from the USB connector is fed via the diode shown above. So I would expect to see 5V minus a Schottky diode drop. However this is then feed back to the LDO regulator and the 3V3 pin on the Nano. This is what can cause the potential damage.

I think from what most others have found, the USB to serial chip on the Nano board it likely to break!!

73, Dave

kirbimir, 2025-02-05:
HI,
Tqankw for answer. Yes, that's exactly what happened . I put non-solated cable in and damaged the serial chip on nano. However, it was not working at all with isolated cable so i tried this. Anyxway, I got the full refund from Alixpress  immediately  so I will use it to purcahse 3rd machine and hope it will work OK . Than I'll program it as you shown to me before and hope for the best. 
I know now not to make the same mistake again.
Thanks a lot 
Regards 
Kirbimir