goshante/ats20_ats_ex #32: Building firmware from source with IDE 1.8 generates smaller hex files than the one for download

AlainTH, 2024-12-31:
Hello all,

I plan to make a small modification to v1.18 to fix the encoder being reversed on my ATS-20+ unit.
To check that my build environment is sane, I first tried building the untouched v1.18 source code with Arduino IDE v1.8.18
I have installed the 3 required libraries in their latest versions: pu2clr/SI4735, Tiny4kOLED, TinyOLED-Fonts
I chose:
Board = Arduino Nano because that's what my ATS-20+ has (does it make a difference for the compile?)
Processor = ATmega328P

The sketch compiles successfully despite a number of non-fatal errors: redefinitions, code that isn't ISO C++ compliant, default arguments, gotos etc.
Final status is:
Sketch uses 30708 bytes (99%) of program storage space. Maximum is 30720 bytes.
Global variables use 1245 bytes (60%) of dynamic memory, leaving 803 bytes for local variables. Maximum is 2048 bytes.

If I do a "Sketch > Export compiled binary" to generate the hex file that I want to compare to the other that is provided for direct download, two are produced:
86389 ATS_EX.ino.eightanaloginputs.hex
85884 ATS_EX.ino.with_bootloader.eightanaloginputs.hex

I presume that the file I want is the one with the bootloader. However it's significantly smaller than the hex file that's offered for download:
89848 ATS_EX_v1.18.hex

How come?
When I compare ATS_EX_v1.18.hex and ATS_EX.ino.with_bootloader.eightanaloginputs.hex, they match until line #1919.

I've also tried setting Processor = ATmega328P (Old Bootloader) in the IDE. In that case, the generated ATS_EX.ino.with_bootloader.eightanaloginputs.hex file has the expected size of  89848 bytes, but the two files are quite different in contents.

Am I pursuing an unreachable goal here? Is the hex file offered for download generated in a different way? (the README says that IDE 1.8 is used though) or from a different source tree? (different options in defs.h maybe?)

G8PTN, 2024-12-31:
Hi,
From my side I have only uploaded the firmware directly from the Arduini IDE, and it seems to work fine.
I have generated the binaries, and they are also smaller than the original. Perhaps this is due to library differences.
If you program via the USB interface, I recommend that the VBUS connection in the USB cable is isolated (see thread below), to avoid any possible contention.
https://github.com/goshante/ats20_ats_ex/issues/19#issuecomment-2361906649
73, Dave

AlainTH, 2025-01-01:
Thanks for your reply, Dave.
Has anyone tried flashing the hex file you have generated? I was nervous enough already when I flashed the "official" v1.18 hex file from Goshante, now I feel even more nervous flashing hex files that comme from the same source code and end up being significantly smaller and different. I fear bricking my ATS20+. I use AVRDUDESS for flashing.
I can't flash from the mini-USB port, hence my buying and using an USBasp programmer. When I connect the ATS-20+ to the PC, it shows up as unrecognized device. Windows does not even attempt to load a driver. This seems to be a common issue on many ATS-20+. Would cutting off the 5V line as suggested make it work?

G8PTN, 2025-01-01:
Hi,
One person has used the hex file I generated after moving the battery monitoring pin from “A2” to “A1”, but this was using AVRDUDESS via the ATS-20+ USB port, not the ISP.

The zip file in the following thread contains the original and modified hex files.
https://github.com/goshante/ats20_ats_ex/issues/19#issuecomment-2380031056
On my ATS-20+ the original bootloader was set to 115200 baud, some appear to be  57600 baud.

On your ATS-20, for whatever reason it sounds like the USB-SERIAL converter may not be working.
If you have a spare USB lead that you can cut the VBUS line, it might be worth trying. When using the isolated USB connection, the ATS-20+ will need to be turned “ON” since the power will be coming from the on-board 3V3 regulator.

Another option is to try connecting a separate UART board to the Nano Serial pins. In this case you would need to press the RST button on Nano when starting to upload, since this puts it Nano into bootloader mode. 

If you have already successfully flashed the Nano via the ISP port using the original hex file, I expect your updated version will also work.

I hope the information helps.
73, Dave

AlainTH, 2025-01-01:
Thanks again Dave,
The mini-USB port not working at all is an issue apparently found on quite a few batches of the ATS-20+.
See this video, that shows how to use the USBasp programmer with the IDE (and not with AVSDUDESS like I did).
https://www.youtube.com/watch?v=7e9yp1MgGI8 
The author mentions this in the beginning.
Flashing this way is much more complex and error-pronoe than when using AVRDUDESS so I'm not quite willing to take the jump yet.
I'm sure I can sacrifice an old mini-USB to USB-A cable and give it a chance. The wire that needs to be cut it the one getting the +5V from the PC, right?
73s, Alain

G8PTN, 2025-01-01:
Hi Alain,
Yes, you should just need to cut the +5V connection in the USB-A to USB-Mini cable.
I agree it is easier to use AVRDUDESS, so stay with that method for now.
I hope you manage to get it programmed with your update.
73, Dave

AlainTH, 2025-01-02:
Hello Dave,
Well, when I cut the +5V (red) wire on a mini-USB to USB-A cable, my ATS-20+ is no longer detected at all by the PC when I connect it. No sound, no notification, nothing at all. I've read somewhere that using a regular cable that causes this power line conflict may fry the USB-to-serial chip on the Arduino. I'm afraid that's what has happened to mine.
So I'm stuck to using my USBasp programmer with AVRDUDESS.

In the meantime if someone can shed some light on these hex files generated by the IDE, which one I should flash (with bootloader?) and why they are so different from the one offered for direct download, you're quite welcome to do so. Thanks in advance.

73s, Alain

G8PTN, 2025-01-02:
Hi Alain,
Sorry to hear that the USB interface does not appear to be working.
Assuming you do not intend to try and fix this, you should be able to flash either hex file via the ISP. The hex file with bootloader is only required if you are using the serial TX/RX signals for uploading, which are connected to the USB-Serial chip.
I am afraid I can’t give a definitive reason for the hex file size differences.
73, Dave