goshante/ats20_ats_ex #35: Is there any way to disable the automatic gain control AGC?

BrahyMasterx, 2025-01-14:
I would like to be able to disable the automatic gain control AGC, is there any way to do it by modifying the code? I have tried, but I have not been able to achieve it.
I would appreciate any help, Thanks.

G8PTN, 2025-01-15:
Hi,
My understanding is that the “ATT” parameter in the setting page is used to control the AGC and manual attenuation.
ATT = AUT		AGC is enabled
ATT = 1 to 37		AGC is disabled, 1 = Min attenuation, 37 = Max attenuation.

Extract from code:
[image]

Extract from PU2CLR documentation:
[image]

I think there may be a couple of minor bugs in the code related to this feature.
1. The generated range for AM/SSB will be 0 to 36. Allowed range is 0 to 37.
2. The generated range for FM will be 0 to 36. Allows range is 0 to 26.

In FM mode, if “ATT” is set > 28, the receiver appears to be in an unknown condition, even if the “ATT” is reset to an in-range value. Toggling between AM and back to FM appears to clear this condition. 

73, Dave

G8PTN, 2025-01-15:
Hi,
I did a  few more tests with serial debug enabled, see details below.

The behaviour seems as expected for AM and FM (apart from out-of-range issue), but for SSB, although the write values seem correct, the read attenuation values are always “0”. I am not sure if this is as expected when the SSB patch is enabled.

73, Dave

[image]

For AM and FM modes, the behaviour is as expected and the manual AGC attenuation seems to work.
Values shown below for several ATT settings.
[image]

For SSB, although the AGC enable/disable appears to be working, the manual attenuation values have no effect.
Values shown below for several ATT settings.
[image]

BrahyMasterx, 2025-01-15:
Thanks for your help
I was trying again, and selecting the value 1 in the AGC does not work, it is as if it activates the maximum gain, generating overmodulation for strong signals, but as I select the other values, the attenuation, which the receiver's own integrated has by default, is clearly noticeable.
But as such, the automatic gain control is not deactivated even if I select the value 1
so there is no way to deactivate the AGC, I am interested in deactivating it for AM

G8PTN, 2025-01-15:
Hi,
As stated, for AM and FM, the manual attenuation does appear to work as expected. If the value is increased in the settings page, the attenuation is increased.
73, Dave

BrahyMasterx, 2025-01-16:
I understand now. Thanks for your help. 73