goshante/ats20_ats_ex #37: Guys can you help me please

ShasaLegend, 2025-01-17:
I want to add in my ats frequency tuning led I saw in the SSM (s meter) settings and there was the rss (rssi) setting. I want the LED to light up when it finds a signal. How can I do that? Where should I connect the LED?

ShasaLegend, 2025-01-17:
I mean i want to solder and lead to the board when it finds a signal led will light up.

G8PTN, 2025-01-18:
Hi,
SSM (SSB Soft Mute Mode) is not an S-Meter setting. The SSM can be based on RSSI or SNR.

If you want to enable the S-Meter, long press the “STEP” button.
The ATS20_EX S-Meter range is 0 to 64 dBuV, where each bar represents 4 dBuV. Example shown below.
[image]

The SI4732 RSSI is defined as shown below.
[image]

I am not sure a single LED would be especially useful, but it could be added with a firmware change and an additional output pin. RSSI thresholds would need to be added to set when to turn the LED on an off.
73, Dave

ShasaLegend, 2025-01-19:
Like to add a led i need to program ats?
And that s meter works fine?

G8PTN, 2025-01-19:
Hi,
As stated previously “SSM” is nothing to do with the S-Meter, it is the SSB Soft Mute option.

The soft mute tries to reduce the so called “chuffing” sound that is heard during SSB tuning. The option selects how the soft mute behaves, and the effect will only be apparent every 32 kHz, since the BFO is used over a 32 kHz span.

If you want to add an LED to indicate when the RSSI level reaches a particular threshold, you will need to modify the code and reprogram the ATS20.
73, Dave

G8PTN, 2025-01-19:
Hi,
As stated previously “SSM” is nothing to do with the S-Meter, it is the SSB Soft Mute option.

The soft mute is intended to improve the listening experience (i.e. reduce static noise). To see any effect, the “SM” will need to be set to a value between 1 and 32. When it is set to 0, soft mute is effectively disabled.

For SSB, the “SSM” option selects whether RSSI or SNR is used for the soft mute mechanism.

If you want to add an LED to indicate when the RSSI level reaches a particular threshold, you will need to modify the code and reprogram the ATS20.
73, Dave

ShasaLegend, 2025-01-19:
Thanks 👍 73

closed 2025-01-19 by ShasaLegend