goshante/ats20_ats_ex #38: Suggestion for radio ATS20+ and battery level presentation

Jedrusia, 2025-02-04:
I have a proposal regarding only the ATS20+ receiver. 
This receiver already has a resistor divider used to determine the battery charge level. It is connected to input A1 of the Arduino Nano board. Can you take this fact into account and produce a HEX version of the file to be uploaded only for the ATS20+ model?
There would be no need to make modifications to the equipment.

Another topic is the bandpassfilter 74-108MHz (available only in the ATS20+ model), so I don't know why to expand the FM band from 64MHz, since the filter will cut signals between 64 and 74MHz anyway. The ATS20 version does not have this filter,

G8PTN, 2025-02-05:
Hi,
The attached zip contains the following hex files.

- ATS_EX_v1.18.hex
With bootloader code (57600 baud)
Original. Needs hardware modification to connect potential divider to Nano "A2" pin.

- ATS_EX_v1_18_A1.hex
No bootloader code.
“A2” to “A1” swap. No hardware change required.

- ATS_EX_v1_18_A1_with_bootloader.hex
With bootloader code (115200 baud)
“A2” to “A1” swap. No hardware change required.

Try the file “**ATS_EX_v1_18_A1.hex**”, which does not contain the bootloader section.
It should be possible to upload via the USB port.

Please note that this is not an official update and should be used at your own risk.

73, Dave

[ATS_EX_hex_230125.zip](https://github.com/user-attachments/files/18675923/ATS_EX_hex_230125.zip)

kirbimir, 2025-02-05:
> From "G8PTN" ***@***.***>
> To "goshante/ats20_ats_ex" ***@***.***>
> Cc "Subscribed" ***@***.***>
> Date 25/02/05 6:04:45 PM
> Subject Re: [goshante/ats20_ats_ex] Suggestion for radio ATS20+ and
> battery level presentation (Issue #38)

Hi,
I could not program it since Win11 and Linux debian did not recognize th 
device on COM serial ports . Win stated that there is an error on the 
device and can not recognize it. Also Arduino IDE could not recognize 
COM port. SO I plugged in standard cable and it works well on stations 
but must be alway connected to 5V . I presume 3V portion did not work 
from the beginning.
However, is there a way to program it with AVRDUDESS using this standard 
cable since with modified cable it does not light up at all.AVRDUDESS 
also does not recognize  COM serial port.

This email has been checked for viruses by Avast antivirus software.
www.avast.com

Jedrusia, 2025-02-05:
I have a similar problem. My ATS20+ radio, after connecting with a USB cable, is detected by the computer as a COM port. And that's where it all ends. An attempt to read the contents of the radio or write to the radio with the AVRDUDESS program ends with an error that the port cannot be opened and therefore nothing can be read or written. To check, I took another Adruino Nano V3 board (new) and connected it with the same cable to the same port and the AVRDUDESS program read and wrote the data to the new board.
Here: https://www.reddit.com/r/HamRadio/comments/1fi16ls/ats_20_ats_ex_firmware_adding_the_battery_level/
I read that you need to cut off the 5V USB power line to be able to upload anything to the Nano V3 on the radio. I haven't checked this solution, can anyone confirm that this is the case?

G8PTN, 2025-02-05:
Hi,
To upload the bootloader version, you will need to use the ISP interface with a programmer such as USBASP or another Arduino (as a programmer).

Most Arduino boards are delivered with a bootloader pre-installed and therefore subsequent uploads can be done via the USB (Serial) interface.

If you upload the bootloader version via USB it will still load the compiled sketch section, but the bootloader section will not be modified. The verification may show a difference  if the pre-installed bootloader code is different to that in hex file. 

Unless you need to program via the ISP interface, I would recommend using the hex file I suggested.

**Unfortunately as stated in many of the threads, connecting the Mini-USB port on the ATS-20+ without isolating the 5V can cause a problem.**

I suggest you have a read though the threads.
73, Dave

G8PTN, 2025-02-06:
Hi,
The baud rate differs between units depending on which bootloader has been installed by the seller.
You could try repeating the test with the baud rate set to 115200 instead of 57600.
73, Dave

va3ngc, 2025-02-06:
Thanks G8PTN - It worked on the first go for me.  Glad I didn't have to modify the board at all (not that I wasn't prepared to do it).

Jedrusia, 2025-02-06:
I don't have a USBASP programmer at hand, but I found a USBISP 2.0 programmer in the resources - a Chinese clone that does not work with AVRDUDESS but works with ProgISP. With this programmer, I uploaded the boot firmware for AST20+, where by default the battery voltage is measured on pin A1.
Everything works as it should. I confirm that the HEX files prepared by my colleague @G8PTN work properly.
Thank you for your help

Itsisme, 2025-04-01:
Used this one : ATS_EX_v1_18_A1_with_bootloader.hex and used AVRDUDESS with 115200 baud updated my ATS-20+ and did a reset (press encoder and switch device on and wait for messages)
After that worked great withou bat mod ...

Thnxs for your efforts...

Chrs..

chaoticanarchyx, 2026-08-29:
In case anyone is still struggling to reflash your device using Linux (Mint) with a USB cable. I used this CLI to be able to reprogram/reflash it successfully.

avrdude -c arduino -p m328p -P /dev/ttyUSB0 -U /home/user/Downloads/ATS_EX_hex_230125/ATS_EX_v1_18_A1.hex

This will change depending on where you store the file. But this command should for the most part get you up and going. You might need to change the ttyUSB0 to whatever yours is assigned to (I.E. ttyUSB1, ttyUSB2)