goshante/ats20_ats_ex #39: Little suggestion for display STEREO symbol on FM

Jedrusia, 2025-02-06:
Would it be possible to show the "Stereo" tag on the display in FM mode, e.g. as a graphic symbol of two letters "o" joined together in half? Something like this:
https://commons.wikimedia.org/wiki/File:Stereo.png

Thanks

diqezit, 2025-04-22:
It is possible, but it requires more code - and the best way is to use the * symbol over the existing SYNC mode position.  
As an option, can draw a graphical primitive ∞. 
But I'm not sure that this library supports this font.

Let's stop at *, need to modify - after adding 

```
#if USE_RDS
    showRDS();
#endif

// Check and update stereo status only if we are in FM, S-meter is off, RDS is not displayed and settings are not active.
if (g_currentMode == FM && !g_sMeterOn && !g_displayRDS && !g_settingsActive) 
{
    g_si4735.getCurrentReceivedSignalQuality(); // Get current signal data
    bool newStereoStatus = g_si4735.getCurrentPilot(); // Check the presence of stereo pilot tone

    // If the status has changed, update the flag and redraw the indicator
    if (g_stereoStatus != newStereoStatus)
    {
        g_stereoStatus = newStereoStatus;
        showModulation(); // Update the modulation display (including the indicator *)
    }
}
```

And replace the function

```
void showModulation()
{
    // Display mode name (FM, AM, SSB, etc.)
    oledPrint(g_bandModeDesc[g_currentMode], 0, 0, DEFAULT_FONT, g_cmdBand && g_currentMode == FM);

    // Do not display additional indicators (S/*) if the settings mode is active
    if (!g_settingsActive) {
        oled.setCursor(24, 0); // Set cursor for S/* indicator

        if (isSSB() && g_Settings[SettingsIndex::Sync].param == 1)
        {
            // Display 'S' for SSB Sync
            oled.print(“S”);
        }
        else if (g_currentMode == FM && g_stereoStatus)
        {
            // Display '*' for FM Stereo
            oled.print(“*”);
        }
        else
        {
            // In other cases (including AM or FM without stereo) print a space to clear the space
            oled.print(” ”);
        }
    } else {
         // If the settings are active, we should also clear the space, if there was something there.
         oled.setCursor(24, 0);
         oled.print(” ”);
    }

    showBandTag(); // Display Bandpass filter / BFO / etc.
}
```

Well so.. Only appears incorrect display of characters on the screen, apparently somewhere where the text buffer is not cleared or incorrect order of rendering. So for now

Jedrusia, 2025-04-22:
Good solution, unfortunately I'm very bad at programming, so it would be great if someone could generate a *BIN file for me with the bootloader and pin A1 set to the battery level indicator.

Thank you.

ColinPamplin, 2025-05-09:
I wonder how you managed to get that code to compile.
I get "text section exceeds available space in board."
Sketch uses 30792 bytes (100%) of program storage space. Maximum is 30720 bytes

diqezit, 2025-05-31:
> I wonder how you managed to get that code to compile.
> I get "text section exceeds available space in board."
> Sketch uses 30792 bytes (100%) of program storage space. Maximum is 30720 bytes

According to the datasheet the ATmega328P physically has 32,768 B of flash memory. A part of this memory is reserved for the bootloader:

The old bootloader occupies 512 B
This leaves 32 256 B for the sketch and the IDE will show something like - Total available 32 256 B

The new bootloader (Optiboot) takes 2,048 B
This leaves 30,720 B for the sketch and the IDE will now show Total 30,720 B available.

If you see “Total available 32,256 B” in the Arduino IDE, it means that the old bootloader (512 B) is used. If you see “Total available 30 720 B” - Optiboot (2 KB) is used - in my case I use the old bootloader which saves memory for flashing

Translated with DeepL.com (free version)