goshante/ats20_ats_ex #41: Perhaps outated informaton wrt Uno vs Nano?

cantbothertoshave, 2025-03-29:
Hello,

I bought this radio weeks ago from AliExpress.
It has the separate board with "Nano" on it. But selecting the lower baud rate does not work.
I used xLoader first, reading someone else's tutorial how to flash this, but it seems, as warned on this site, the USB power was not enougzh and it only "half" worked. There is somewhat working firmware on it. Is the ~ 2x loud click after the boot screen showed the verison and it transitions to the operational screen normal, or already hints at a problem?
I can receive some stuff... so *something* works.
But if I try to reflash or verify, now with the much nicer AVRDUDESS, I can, again, only flash with the higher baud rate (the lower gets my sync errors), but at the end it reads, and the verify always fails: **device 0xff != input 0x0c at addr 0x7800**. I did *not* activate the "disable flash erase" option.

I also tried the Arduino IDE as suggested - here choosing the old bootloader makes everything fail, and the new one makes uploading a sketch work.

Flashing again the v1.18.hex results in the same mismatch error quited above, and it's still clicking hard after the boot screen.
I downloaded the firmware on a 2nd computer and compared in HxD hex editor the one I flashed vs. the newly downloaded - they are identical.

So, if anything, my post can serve to show that there are now "nano" ones where "old bootloader" and lower baud rate are wrong,

It would be great if someone could help me find out why I am getting the flash verification mismatch, though...
Could they have used fake or damaged arduino in my specimen?

G8PTN, 2025-03-29:
Hi,
The verification warning/error is not necessarily a problem, it is probably because the Nano has different bootloader code to that contained in the hex file you are uploading.

When uploading via USB (serial port), the bootloader will not be overwritten, but the verification will compare all content defined in the hex file with what it reads from the Nano.

Some Nano’s ship with the 57600 bootloader and others the 115200 version.

Please see the following link. You could try loading the ATS_EX_v1_18_A1.hex file which has the “A2” to “A1” swap and the battery meter should then work. No hardware change required.
https://github.com/goshante/ats20_ats_ex/issues/38#issuecomment-2637517788

**It should be noted that connecting directly to the USB port without isolating the VBUS (5V) could damage the Nano. An explanation can be found in the following link.
https://groups.io/g/si47xx/message/842**

Unfortunately, the loud clicking sound on boot up is normal with the ATS-20+ and similar receivers using the SI4732 receiver chip. You will also find switching between FM/AM/SSB will produce the clicking sound.

I hope the information is of some help.

73, Dave

Nek0-San, 2025-07-26:
Hey guys. On the v2 version of the 20+ there is no Arduino nano pun a1 and a2. Any ideas?