goshante/ats20_ats_ex #42: LGT8F328P vs ATmega328P - potential processor upgrade & replacement in ATS-20

diqezit, 2025-04-19:
# Comparing ATmega328P vs LGT8F328P for SI4732 

As an alternative to the Arduino Nano 328P, the LGT8F328P microcontroller is gaining more and more attention
It remains instruction-compatible with the ATmega328P, and provides slightly higher performance with several advantages for battery-powered applications

## 1. Arduino Nano 328P Overview
[image]
- **MCU:** ATmega328P (8‑bit AVR)  
- **Clock:** 16 MHz (~16 MIPS)  
- **Memory:**  
  - Flash: 32 KB (2 KB for bootloader)  
  - SRAM: 2 KB  
  - EEPROM: 1 KB  
- **I/O:**  
  - 14 digital pins (6 PWM)  
  - 8 analog inputs (10‑bit, 1024 levels)  
- **Interfaces:** UART, I²C, SPI  
- **Power:** 5 V (recommended 7–12 V, accepts 6–20 V)  
- **Extras:** Onboard LED on D13, extensive documentation and code examples

## 2. LGT8F328P Overview
[image]
- **MCU:** 8‑bit RISC‑AVR core, AVR‑instruction compatible  
- **Clock:**  
  - Internal up to 32 MHz  
  - External crystal 16 MHz supported  
- **Memory:**  
  - Flash: 32 KB  
  - SRAM: 2 KB  
  - EEPROM: emulated in Flash (configurable 1–8 KB, reduces available Flash)  
- **I/O:**  
  - 14 digital pins (6 PWM, plus extra GPIO on some packages)  
  - 8 analog inputs (12‑bit, 4096 levels)  
- **Voltage:** 1.8–5.5 V  
- **Extras:**  
  - 8‑bit DAC  
  - Internal references: 1.024 V, 2.048 V, 4.096 V  
  - Signal amplifier for ADC  
  - Extra timers (Timer3) with up to 9 PWM channels  
  - Unique ID (GUID)  
  - DSC accelerator for fast 16‑bit math  
  - PWM dead‑time control  
  - Some boards add LDO, USB‑C, Bluetooth 5.0, hardware encryption

## 3. Performance Comparison
- **Clock speed:** LGT8F328P can run at twice the frequency (32 vs. 16 MHz)  
- **Benchmarks:** At 32 MHz, LGT8F328P completes simple loops almost twice as fast as a Nano at 16 MHz  
- **DSC accelerator:** Speeds up 16‑bit operations further  
- **Setup note:** You must adjust clock prescalers to actually reach 32 MHz

## 4. Memory & EEPROM
- **Flash/SRAM** are identical: 32 KB / 2 KB  
- **EEPROM:**  
  - Nano has dedicated 1 KB  
  - LGT8F328P emulates EEPROM in Flash (≤8 KB, wear limited to ~100,000 cycles, uses part of Flash)  
- **Reliability:** Dedicated EEPROM generally endures more cycles per byte than emulation

## 5. Other Key Differences
- **Voltage range:** LGT8F328P: 1.8–5.5 V (vs. 5 V on Nano)  
- **ADC accuracy:** 12‑bit on LGT8F328P vs. 10‑bit on Nano  
- **DAC & references:** Only available on LGT8F328P  
- **Pin drive:** Up to 80 mA on some LGT8F328P pins  
- **Programming:** Some LGT8F328P boards use SWD/SWC ISP, and may need manual reset before upload  
- **Documentation:** Nano has complete, well‑translated docs; LGT8F328P docs can be sparse or in Chinese

## 6. Pin Compatibility Comparison
[image]

- LGT8F328P can effectively control the shutdown mode of MD8002A (pin 1 in the schematic)
- Approximately 1.5× longer battery life while maintaining all functions
-  SI4732 uses a 3.3V I²C/SPI interface, and the LGT8F328P can work directly with this voltage without level converters
- Additional LGT8F328P interrupts allow more efficient processing of SI4732 events (RDS data, signal level changes)

With almost complete software compatibility and minimal changes to existing circuitry, this replacement appears highly beneficial for portable SI4732 radio projects where power efficiency is critical.

diqezit, 2025-04-20:
with certainty I can say that the modification with the connection of pin 1 to the arduino pin and modification of the firmware remained **quite working** - I lost all the impulse jumps when switching the internal amplifier si4732 - you just need to modify a little bit of firmware in some places

droneando, 2025-05-12:
Hi diqezit: I have implemented your audio pops fix. It works fine but the display also goes down every time the md8002a does, and come back to life with the amplifier. I dont know if it is normal behaviour or any thing goes wrong with my fix...

diqezit, 2025-05-14:
> Hi diqezit: I have implemented your audio pops fix. It works fine but the display also goes down every time the md8002a does, and come back to life with the amplifier. I dont know if it is normal behaviour or any thing goes wrong with my fix...

Try to reset the memory eeprom after flashing

droneando, 2025-05-14:
As soon as I flashed the Arduino I reseted the eeprom, but I'll reset it again

jh4vaj, 2025-06-24:
Thank you, I gave it a try. 
Since the code won’t compile under the Nano board setting (it throws a compile error), 
I switched to the UNO setting instead. 
My Nano uses the Optiboot bootloader, so I believe using the UNO configuration is fine
—in fact, that's how I've always built it without issues.

However, in this test, the ATS-20+ ends up stuck in a reboot loop. 
It might be that the binary is too large and corrupting memory.
I've uploaded a video to YouTube showing the reboot behavior (unlisted video).

https://youtu.be/Vv14DVetGHw

jh4vaj, 2025-06-24:
Apologies — after downloading and compiling the latest MOD_NO_RDS firmware, the ATS-20+ now boots up correctly. SSB demodulation also seems to be working fine.

Please note that I'm using the Arduino Uno board setting. It wouldn't compile under the Nano setting (compile error occurred).

Thanks again for the quick fix.

P.S. I really appreciate that you integrated my modification for the 9 kHz step issue.

[image]

jh4vaj, 2025-06-30:
I noticed that a new version of MOD_NO_RDS has been released—its progress is truly impressive.

On the MW band, I encountered an unexpected issue.

During reception, audio from a specific station broadcasting at 954 kHz is mixed in, no matter which station I tune into. 
Even when receiving 594 kHz, 693 kHz, or 810 kHz, the 954 kHz audio can always be heard simultaneously.

When tuned to 954 kHz, the audio is severely distorted.

This issue occurs only when using a small loop antenna (donut antenna). 
It does not happen when using a telescopic antenna or a larger external antenna.

I was able to receive signals normally with the goshante firmware or earlier versions of MOD_NO_RDS. 
I haven't tested MOD_NO_RDS v2, v3, or v3.1—only v3.2 I tested. 
The comment at the beginning of the v3.2 source code still indicates "v3.1," so I'm unsure whether it truly is v3.2.

[image]

diqezit, 2025-06-30:
> On the MW band, I encountered an unexpected issue.
>
> During reception, audio from a specific station broadcasting at 954 kHz is mixed in, no matter which station I tune into. Even when receiving 594 [...]

This was a problem in version 3.2 starting in v3, when I made experimental changes to the ATT settings
I wanted to remove the AUTO mode to reduce background noise, but it inevitably made the receiver run at maximum sensitivity, which caused the very overload problem you describe, especially in areas with strong MW stations

In the new version v3.3 I have restored AUTO mode as a setting as originally provided in the original firmware so this should fix the problem