goshante/ats20_ats_ex #43: Display some "special" Mode like Airband with an Down Convverter + Channel Name?

nousername1, 2025-06-03:
Hi
I read the "readme" but I suggest something. I saw online there is an "external" Down Converter for that Device who allow us to listen to Airband with the Radio. To get the "full experience" it would be nice when I could add the "fix" offset into the Radio when I would select "Airband" for example.

Another thing who would be nice is there any way to load an Station list into the Radio to quick Tune known Radio and other Station in.
I dont know much about the device since the are still on the way from China to my Place but when Memory of the Nano (?) or what ever is solder in the device is a problem is there a way to expand them by solder an other in or maybe add another memory chip?
Thanks for the patient with me.

nousername1, 2025-06-19:
Sorry when I ask some other stupid question!
Is it potential possible to tune to a certain frequency from outside of the Case maybe via a 2 Wire Connection (in the Future/ different FW, SW)?
The reason is I think about building an SDR Style Receiver with Multiple Radio Chips who each one only Monitor 1 Freq 24/7.
For that I (= a PCB Designer/ Engineer) would remove everything who is not need for the reception from the current PCB like the Display, Physical Button (maybe keep a "reset" one or so), HW Knob to tune everything remotely.

Does anyone here know more about the Radio Chip? 
" and DOUT (pin 17) for digital audio mode. "
But how does the Digital Audio work? Like on the other End who kind of Chip do I need to allow an OS to see them as an Audio Source?
"DIGITAL AUDIO (Si4735)"
https://d1ehax0mqsd4rz.cloudfront.net/-/media/SkyWorks/SL/documents/public/data-shorts/Si4734-35-short.pdf
Thanks!

G8PTN, 2025-06-19:
Hi,
No problem with the questions.

It would be possible to have a headless version and there are some examples in the PU2CLR library.
https://github.com/pu2clr/SI4735

An  MCU would still be required for the low-level control of the SI4732-A10 via the I2C interface, and again there are several examples in the PU2CLR library.

The audio (analogue) would either need to be amplified for headphone/speaker outputs or fed into a suitable ADC and processed/streamed. Perhaps a standard USB soundcard dongle or module could be used if connecting to a PC.

With respect to the remote control, there are many options depending on the MCU and application. These could include USB (Serial), Bluetooth (Serial), WIFI (Webserver) and probably many more.

A few application notes can be found on the Internet for the SI473x family of devices.
AN332 - Si47XX PROGRAMMING GUIDE
AN332 (SSB) - AMENDMENT FOR SI4735-D60 SSB AND NBFM PATCHES
AN383 - Si47XX ANTENNA, SCHEMATIC, LAYOUT, AND DESIGN GUIDELINES

For the ATS20+ with the Arduino Nano, I added a simple Bluetooth to Serial module to allow basic control. Example shown in the link below.
https://github.com/goshante/ats20_ats_ex/issues/7#issuecomment-2308964134

I have also experimented with the SI4732 Mini Receiver hardware, which uses the ESP32-S3 as the MCU. There is lots of firmware development activity for this hardware at the moment and you could look at the following GitHub site.
https://github.com/esp32-si4732/ats-mini

The ats-mini firmware allows basic serial control over USB, but I believe that full CAT style control over USB or Bluetooth is being considered. There is also limited control over WIFI with a Webserver on the ESP32-S3.

The ESP32 has the capability to perform decoding for Morse, RTTY, FT8 and other digital modes, as per the ATS-25 receivers that are already available.

The SI4732-A10 is a great little device and there are endless options. There will always be performance limitations, some of which can be mitigated with external filtering and so. You should not expect too much from the device, but it's great for experimentation and learning.

For headless receivers, you could also consider SDR based options such as the RTL-SDR dongles or more advanced hardware like the RSP1A, some of which allow multiple slices (receivers) using the same hardware. 

Sorry, if I have overloaded you with too much information, perhaps some of it may be useful !!

73, Dave

Additional Notes:
Some explanations below for the SI4732-A10 device.

The SI4732-A10 receiver device is used in many low-cost radios, including the ATS-20+ and others such as the ATS-25.
The device was developed many years ago, and I believe it was primarily intended for small portable LW/MW/SW (AM) and VHF (Broadcast FM) applications.

[image]

Since the receiver implements the audio demodulation using digital signal processing (DSP) , a firmware patch was developed (source unknown) which provides SSB demodulation.

Control:
The SI4732 is controlled via an interface called “I2C” (Integrated-Integrated Circuit), which was developed to allow communication between microcontrollers and peripheral devices. In the case of the ATS-20+ the MCU is the ATMEGA328P(PB) on the Arduino Nano module and the peripheral device is the SI4732-A10.

The Arduino Nano writes to registers inside the SI4732-A10 to set frequency, modulation, bandwidth and so on. It then reads back from other registers to get status information such as RSSI, SNR.

For SSB, the firmware patch code has to be first downloaded via the I2C interface, and then run in the embedded processor on the SI4732-A10 to perform the SSB demodulation functions.

Audio:
The demodulated audio is normally fed via the digital to analogue (DAC) blocks in the SI4732-A10 to produce the LOUT and ROUT analogue signals. These outputs are at a low level and need an to be amplified before feeding to headphones and/or speaker.

There is an option to output the audio in digital format called “I2S” (Inter-Integrated Circuit Sound), which was designed to transmit digital audio between integrated circuits.

There are many devices that can support I2S, including Codecs, Microphones, Amplifiers and MCUs. When used with the correct type of MCU or processor, the digital audio can be modified (filtered, equalised etc) and streamed over interfaces such as USB, Ethernet, WIFI and Bluetooth.

In the case of the SI4732-A10, it can only be used in either analogue or digital audio mode. For SSB, when the patch is loaded, digital audio is not possible.

G8PTN, 2025-06-19:
Hi,
If all the SI4732-A10 receivers are on the same PCB and only need to be setup one by one from the same MCU, then a simple I2C multiplexer module could be used, such as the TCA9548A. Polling information from the receivers could be done on a cyclic basis via the multiplexer.

Using I2S for the audio would mean that SSB reception would not be possible, and you would need to check which versions of the SI473x support digital audio. Some reports suggest that the SI4732-A10 does not work.

If you are going to connect everything up to a PC, then I would have thought RTL-SDR dongle(s) would be easier, since the control and IQ data is over the same USB interface.

The frequency range on the SI4732-A10 is 150kHz to 30MHz (AM/SSB with patch) and 64MHz to 108MHz (Broadcast FM).
RTL-SDR dongles can cover something like 100kHz to 1.75GHz and any mode can be demodulated from the IQ data.

73, Dave

G8PTN, 2025-11-09:
> 3\. My question is different - how hard would be to prepare Squelch functionality based on Si4732 signal strenght informations?

Hi,
It would be possible to implement this with a firmware change.

For information, it is already implemented in the ats-mini firmware which you could use for reference.
Path: https://github.com/esp32-si4732/ats-mini

[image]

73, Dave

ormopl, 2025-11-09:
Dave, 
1. I have seen such an converters, i used to build my own, back in the days, especially for CB TRX'es :)
2.  Thank you for the link for ATS Mini Software - i will take a look on it.
3. Yes, i have couple of them! None of them is working properly on airband - AGC is too weak for Airband Operation :)