goshante/ats20_ats_ex #44: Scan Behavior When Using the 9 kHz Step

jh4vaj, 2025-06-06:
When the frequency step is set to 9 kHz, the scan does not work properly. 

For example, if the scan starts at 810 kHz and is manually stopped, 
manually tuning the frequency afterward results in values like 808 kHz or 817 kHz instead of returning to 810 kHz. 

However, when starting the scan from the lower band limit of 520 kHz, it functions correctly. 
This suggests that the frequency is rounded to 9 kHz intervals starting from the lower band limit (9 × 32 + 520 = 808). 

This issue only affects scanning, so it is not a major concern. However, I am reporting it.

jh4vaj, 2025-06-12:
I came up with a simple workaround for this issue. 
Setting the lower limit of the MW band to a common multiple of both 9 kHz and 10 kHz
—namely, 450 kHz—should make the behavior consistent, whether the step is 9 kHz or 10 kHz.

Although this extends the tuning range unnecessarily low, it should have no other side effects. 
Since it only requires redefining the band’s lower limit, the binary size will not increase.

Specifically, the MW band definition in globals.h can be changed as follows:

```
Band g_bandList[] =
{
    /* LW */ { LW_LIMIT_LOW, 520, 300, 0, 4 },
    ///* MW */ { 520, 1710, 1476, 3, 4 },
    /* MW */ { 450, 1710, 1080, 3, 4 },
    /* SW */ { SW_LIMIT_LOW, SW_LIMIT_HIGH, SW_LIMIT_LOW, 0, 4 },
    /* FM */ { 6400, 10800, 8400, 1, 0 },
};
```

I also changed the MW default frequency to 1080 kHz, which is a common multiple of 9 and 10. 
The upper limit of 1710 kHz remains unchanged, as it already satisfies this condition.

G8PTN, 2025-06-18:
Hi,
Yes, there are some limitations with the search and step alignment.
It seems that you have an acceptable workaround, so that’s great.

I think there are also some limitations with the manual tuning in terms of it aligning to the selected step boundary. Again this is a minor issue and does not really cause any major problems.

73, Dave