goshante/ats20_ats_ex #51: Battery display for the v2 second version with Arduino embeded in the mainboard

Nek0-San, 2025-07-26:
Hi guys
I have the v2 of theats 20+, the one with no arduino separated board.
Anyone can help sharing info for how to install a voltage divider or if there is a fw mod for this version ?
Thank you

G8PTN, 2025-07-28:
Hi,
This may be covered in a previous thread.
https://github.com/goshante/ats20_ats_ex/issues/31#issuecomment-2559888392

In this case the board had a dual footprint and the 28 VQFN device appeared to be fitted. The user loaded the firmware with the "A1" pin for the battery monitoring.
[image]

**Summary**
- Add a potential divider between the switched battery voltage and the ADC pin on the ATmega328. An example is shown below.
[image]

- Connect the potential divider to the appropriate pin on the ATmega32. The connections options for A1 and A2 are shown below. It might be easier to connect to "A2" on 28 VQFN device. The PCB does not appear to have bridges between all pins on the dual footprint, so the connection will need to be made directly to the associated pin.
[image]

- Program the device with the appropriate goshante firmware, depending on which pin is used. This is covered in the attached thread.
https://github.com/goshante/ats20_ats_ex/issues/38#issuecomment-2637517788

Kind Regards,
 Dave

Nek0-San, 2025-07-28:
> #31 (comment)
>    #38 (comment)

Thank you. I'll keep you informed.